platformio/platformio-core #1183: Home: Could not load recent projects

Cinezaster, 2017-11-13:
Server error

ivankravets, 2017-11-16:
What is your OS?

Cinezaster, 2017-11-16:
OSX

ivankravets, 2017-11-16:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

Cinezaster, 2017-11-17:
```
$ echo $PATH
/Library/Frameworks/Python.framework/Versions/3.6/bin:/usr/local/bin:/usr/local/mysql/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin:/usr/local/MacGPG2/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Users/cinezaster/.platformio/penv/bin:/Library/Frameworks/Python.framework/Versions/3.6/bin:/usr/local/mysql/bin:/Users/cinezaster/Library/Android/sdk/tools:/Users/cinezaster/Library/Android/sdk/platform-toolsPATH=/usr/local/bin:/Users/cinezaster/.rvm/bin:/Users/cinezaster/Library/Android/sdk/tools:/Users/cinezaster/Library/Android/sdk/platform-toolsPATH=/usr/local/bin:/usr/local/mysql/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin:/usr/local/MacGPG2/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Users/cinezaster/.platformio/penv/bin:/Library/Frameworks/Python.framework/Versions/3.6/bin:/usr/local/mysql/bin:/Users/cinezaster/Library/Android/sdk/tools:/Users/cinezaster/Library/Android/sdk/platform-toolsPATH=/usr/local/bin:/Users/cinezaster/.rvm/bin:/usr/local/bin
```

```
$ echo $PYTHONPATH

```

```
$ python --version
Python 2.7.10
$ python3 --version
Python 3.6.3
```

```
$ pip --version
pip 9.0.1 from /Library/Python/2.7/site-packages (python 2.7)
```

```
$ pip list
altgraph (0.10.2)
argh (0.26.1)
Babel (2.1.1)
backports.ssl-match-hostname (3.4.0.2)
bdist-mpkg (0.5.0)
blinker (1.4)
bonjour-py (0.3)
bottle (0.12.13)
certifi (2015.11.20)
ecdsa (0.13)
esptool (2.1)
Flask (0.10.1)
Flask-Assets (0.10)
Flask-Babel (0.9)
Flask-Login (0.2.2)
Flask-Principal (0.3.5)
itsdangerous (0.24)
Jinja2 (2.8)
macholib (1.5.1)
MarkupSafe (0.23)
matplotlib (1.3.1)
modulegraph (0.10.4)
netaddr (0.7.18)
netifaces (0.10.4)
numpy (1.8.0rc1)
OctoPrint (1.2.2-459-g5c8d491)
pathtools (0.1.2)
pip (9.0.1)
pkginfo (1.2.1)
py2app (0.7.3)
pyaes (1.6.1)
pyasn1 (0.1.9)
pylru (1.0.9)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
pyserial (2.7)
python-dateutil (1.5)
pytz (2013.7)
PyYAML (3.10)
requests (2.14.2)
rsa (3.2.3)
Rtree (0.8.2)
sarge (0.1.4)
scipy (0.13.0b1)
semantic-version (2.4.2)
setuptools (18.5)
Shapely (1.5.8)
simplejson (3.6.5)
six (1.4.1)
sockjs-tornado (1.0.2)
speaklater (1.3)
tornado (4.0.1)
vboxapi (1.0)
watchdog (0.8.3)
webassets (0.11.1)
Werkzeug (0.8.3)
xattr (0.6.4)
zope.interface (4.1.1)
```

ivankravets, 2017-11-24:
Could you reproduce this issue when restart IDE?

Cinezaster, 2017-11-24:
Yes

ivankravets, 2017-11-24:
Try to start PIO Home directly via
```
pio home
```

Do you see any errors?

Cinezaster, 2017-11-25:
I see the same error

ivankravets, 2017-11-25:
I mean in the terminal/console.

Cinezaster, 2017-11-25:
```
PlatformIO Plus (https://pioplus.com) v0.10.14

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________

Unhandled Error
Traceback (most recent call last):
  File "/Users/cinezaster/.platformio/packages/pysite-pioplus/txjason/service.py", line 472, in _handle_request
    d = self._call_method(request)
  File "/Users/cinezaster/.platformio/packages/pysite-pioplus/twisted/internet/defer.py", line 1445, in unwindGenera
tor

    return _inlineCallbacks(None, gen, Deferred())
  File "/Users/cinezaster/.platformio/packages/pysite-pioplus/twisted/internet/defer.py", line 1299, in _inlineCallb
acks
    result = g.send(result)
  File "/Users/cinezaster/.platformio/packages/pysite-pioplus/txjason/service.py", line 438, in _call_method
    result = yield defer.maybeDeferred(method, *params)
--- <exception caught here> ---
  File "/Users/cinezaster/.platformio/packages/pysite-pioplus/twisted/internet/defer.py", line 150, in maybeDeferred
    result = f(*args, **kw)
  File "cmd_home.py", line 134, in cmd_home.ProjectRPC.get_projects

  File "/Users/cinezaster/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 181, i
n board_config
    for manifest in self.get_installed_boards():
  File "/Users/cinezaster/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 156, i
n get_installed_boards
    p = PlatformFactory.newPlatform(manifest['__pkg_dir'])
  File "/Users/cinezaster/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 238, i
n newPlatform
_instance = platform_cls(join(platform_dir, "platform.json"))
  File "/Users/cinezaster/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 425, i
n __init__
    self.engines['platformio']):
  File "build/bdist.macosx-10.11-intel/egg/semantic_version/base.py", line 460, in __init__

  File "build/bdist.macosx-10.11-intel/egg/semantic_version/base.py", line 466, in parse

  File "build/bdist.macosx-10.11-intel/egg/semantic_version/base.py", line 466, in <genexpr>

  File "build/bdist.macosx-10.11-intel/egg/semantic_version/base.py", line 404, in __init__

  File "build/bdist.macosx-10.11-intel/egg/semantic_version/base.py", line 419, in parse

exceptions.ValueError: Invalid requirement specification: u'^3.0.0'
```

ivankravets, 2017-11-25:
Please provide an output of
```
pio platform list --json-output
```

Cinezaster, 2017-11-25:
```
[
   {
      "repository":"https://github.com/platformio/platform-atmelavr.git",
      "license":"Apache-2.0",
      "title":"Atmel AVR",
      "url":"http://www.atmel.com/products/microcontrollers/avr/default.aspx",
      "__pkg_dir":"/Users/cinezaster/.platformio/platforms/atmelavr",
      "description":"Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new ones-they are based on the industrys most code-efficient architecture for C and assembly programming.",
      "frameworks":[
         "arduino",
         "simba"
      ],
      "version":"1.7.1",
      "packages":[
         "toolchain-atmelavr",
         "framework-arduinoavr",
         "framework-simba",
         "tool-avrdude",
         "tool-micronucleus"
      ],
      "homepage":"http://platformio.org/platforms/atmelavr",
      "forDesktop":false,
      "name":"atmelavr"
   },
   {
      "repository":"https://github.com/platformio/platform-espressif32.git",
      "license":"Apache-2.0",
      "title":"Espressif 32",
      "url":"https://espressif.com/",
      "__pkg_dir":"/Users/cinezaster/.platformio/platforms/espressif32",
      "description":"Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.",
      "frameworks":[
         "arduino",
         "espidf",
         "pumbaa",
         "simba"
      ],
      "version":"0.10.0",
      "packages":[
         "toolchain-xtensa32",
         "framework-simba",
         "framework-arduinoespressif32",
         "framework-pumbaa",
         "tool-espotapy",
         "framework-espidf",
         "tool-esptoolpy"
      ],
      "homepage":"http://platformio.org/platforms/espressif32",
      "forDesktop":false,
      "name":"espressif32"
   },
   {
      "repository":"https://github.com/platformio/platform-espressif8266.git",
      "license":"Apache-2.0",
      "title":"Espressif 8266",
      "url":"https://espressif.com/",
      "__pkg_dir":"/Users/cinezaster/.platformio/platforms/espressif8266",
      "description":"Espressif Systems is a privately held fabless semiconductor company. They provide wireless communications and Wi-Fi chips which are widely used in mobile devices and the Internet of Things applications.",
      "frameworks":[
         "arduino",
         "esp8266-rtos-sdk",
         "simba"
      ],
      "version":"1.4.0",
      "packages":[
         "framework-esp8266-rtos-sdk",
         "toolchain-xtensa",
         "framework-simba",
         "tool-esptool",
         "tool-mkspiffs",
         "tool-espotapy",
         "framework-arduinoespressif8266",
         "sdk-esp8266"
      ],
      "homepage":"http://platformio.org/platforms/espressif8266",
      "forDesktop":false,
      "name":"espressif8266"
   },
   {
      "repository":"https://github.com/platformio/platform-teensy.git",
      "license":"Apache-2.0",
      "title":"Teensy",
      "url":"https://www.pjrc.com/teensy",
      "__pkg_dir":"/Users/cinezaster/.platformio/platforms/teensy",
      "description":"Teensy is a complete USB-based microcontroller development system, in a very small footprint, capable of implementing many types of projects. All programming is done via the USB port. No special programmer is needed, only a standard USB cable and a PC or Macintosh with a USB port.",
      "frameworks":[
         "arduino",
         "mbed"
      ],
      "version":"2.7.0",
      "packages":[
         "toolchain-atmelavr",
         "framework-arduinoteensy",
         "tool-teensy",
         "toolchain-gccarmnoneeabi",
         "framework-mbed"
      ],
      "homepage":"http://platformio.org/platforms/teensy",
      "forDesktop":false,
      "name":"teensy"
   }
]
```

ivankravets, 2017-11-26:
Could you remove `/Users/cinezaster/.platformio/platforms` folder and restart IDE?

Cinezaster, 2017-11-28:
I removed folder and restarted Atom. Still the same.

ivankravets, 2017-12-01:
Did you receive my email?

ivankravets, 2017-12-03:
Should be fixed in 3.5.0rc5. Please re-test

Cinezaster, 2017-12-04:
I did receive your email. I was caught up by some job.

What I did today. 
1. Removed PlatformIO from ATOM by clicking on uninstall on all three packages. 
2. I removed .platformio folder.
3. I installed platformio ide again via atom package installer. (got one error: "[spawn platformio ENOENT](https://github.com/platformio/platformio-atom-ide/issues/1209)") after that I restarted ATOM 
4. On loading got [this error](https://github.com/platformio/platformio-atom-ide/issues/1229) 
5. Open project selected an existing project
6. Same error again: "Home: Could not load recent projects"

If you want send me again an invite for teamviewer

Cinezaster, 2017-12-05:
Got it working.
Removed python, python3 and atom from my macbook. Also all .files in my user folder.
Reinstalled atom and platformio. Now I have no errors anymore.
Thanks

Cinezaster, 2017-12-05:
To fast, I used the button 'open project' and added an older project and the error appeared again.

ivankravets, 2017-12-05:
Do you have some non-ascii chars in project path?

Cinezaster, 2017-12-05:
I don't think so.
/Users/cinezaster/Documents/PlatformIO/Projects/Influencair_LoraWan_HPM

ivankravets, 2017-12-05:
Could you mail me@ikravets.com and I will provide further steps how to debug this issue?

ivankravets, 2017-12-05:
The problem was linked with broken `semantic-version` Python's library. This command solved the problem.

```
sudo pip uninstall semantic-version
```

P.S: Thanks @Cinezaster for your time!

closed 2017-12-05 by ivankravets