platformio/platformio-core #1184: Home: Could not load boards list

mogplus8, 2017-11-13:
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 52 (char 142001 - 142101)
Found the problem. There is a file called /etc/host-conf, which has two lines in it, "nospoof on" and "spoofalert on". I commented these two lines out and now it all works fine. I don't know what the file does. There are only two other lines in it, "order hosts,bind" and "multi on". So I guess I'll have to wait and see if anything else odd happens... 
;-) Ian

ivankravets, 2017-11-16:
Do you see any errors when you type `pio boards --json-output` in PIO IDE Terminal?

ivankravets, 2017-11-26:
Duplicate of https://github.com/platformio/platformio-core/issues/1164

closed 2017-11-26 by ivankravets