platformio/platformio-core #1185: Home: Could not load lastest application state

chall3ng3r, 2017-11-14:
Still same error, VS extension stuck on "Verifying PlatformIO Core Installation..."

ivankravets, 2017-11-16:
What is your OS?

chall3ng3r, 2017-11-16:
It's Windows 10 (16299.19), and I have to kill python processes on start of VSCode. If I don't do that, VSCode keeps using 50% CPU and "Verifying" state indefinitely.

chall3ng3r, 2017-11-16:
Here's the output for requested commands:

```
echo %PATH%
C:\Users\chall3ng3r\.platformio\penv\Scripts;C:\Program Files\Docker\Docker\Resources\bin;C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Users\chall3ng3r\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\Microsoft Emulator Manager\1.0\;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\MinGW\bin;C:\MinGW\msys\1.0\bin;C:\Program Files\LLVM\bin;C:\Program Files\dotnet\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Users\chall3ng3r\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Microsoft VS Code\bin;C:\Users\chall3ng3r\AppData\Local\atom\bin;C:\Users\chall3ng3r\AppData\Local\Fusetools\Fuse\App\Bin;C:\Users\chall3ng3r\bin;C:\Users\chall3ng3r\AppData\Local\Microsoft\WindowsApps;C:\Users\chall3ng3r\AppData\Roaming\npm;C:\Program Files\Docker\Docker\Resources\bin;C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Users\chall3ng3r\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\Microsoft Emulator Manager\1.0\;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\MinGW\bin;C:\MinGW\msys\1.0\bin;C:\Program Files\LLVM\bin;C:\Program Files\dotnet\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Users\chall3ng3r\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Microsoft VS Code\bin;C:\Users\chall3ng3r\AppData\Local\atom\bin;C:\Users\chall3ng3r\AppData\Local\Fusetools\Fuse\App\Bin;C:\Users\chall3ng3r\bin;C:\Users\chall3ng3r\AppData\Local\Microsoft\WindowsApps;C:\Users\chall3ng3r\AppData\Roaming\npm
```

```
echo %PYTHONPATH%
%PYTHONPATH%
```

```
pip --version
pip 9.0.1 from c:\users\chall3ng3r\.platformio\penv\lib\site-packages (python 2.7)
```

```
pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
arrow (0.10.0)
bottle (0.12.13)
certifi (2017.7.27.1)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
idna (2.6)
lockfile (0.12.2)
pip (9.0.1)
platformio (3.5.0b3)
pyserial (3.4)
python-dateutil (2.6.1)
requests (2.18.4)
semantic-version (2.6.0)
setuptools (36.2.7)
six (1.11.0)
urllib3 (1.22)
wheel (0.30.0a0)
```

ivankravets, 2017-11-16:
Could provide the same output from SYSTEM `cmd` terminal, not IDE.

chall3ng3r, 2017-11-16:
Sorry, here's out from cmd:

```
echo %PATH%
C:\Program Files\Docker\Docker\Resources\bin;C:\Python27\;C:\Python27\Scripts;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Users\chall3ng3r\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\Microsoft Emulator Manager\1.0\;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\MinGW\bin;C:\MinGW\msys\1.0\bin;C:\Program Files\LLVM\bin;C:\Program Files\dotnet\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Users\chall3ng3r\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Microsoft VS Code\bin;C:\Users\chall3ng3r\AppData\Local\atom\bin;C:\Users\chall3ng3r\AppData\Local\Fusetools\Fuse\App\Bin;C:\Users\chall3ng3r\bin;C:\Users\chall3ng3r\AppData\Local\Microsoft\WindowsApps;C:\Users\chall3ng3r\AppData\Roaming\npm;
```

```
echo %PYTHONPATH%
%PYTHONPATH%
```

```
python --version
Python 2.7.9
```

```
pip --version
pip 1.5.6 from C:\Python27\lib\site-packages (python 2.7)
```

pip --version
pip 1.5.6 from C:\Python27\lib\site-packages (python 2.7)

```
pip list
arrow (0.10.0)
bottle (0.12.13)
click (5.1)
colorama (0.3.7)
curl (2017.0.1)
lockfile (0.12.2)
pip (1.5.6)
pyserial (3.2.1)
python-dateutil (2.6.1)
requests (2.12.4)
semantic-version (2.6.0)
setuptools (7.0)
six (1.10.0)
```

ivankravets, 2017-11-16:
Please 

1. Uninstall Python from a system
2. Remove `C:\Python27` and `C:\Users\chall3ng3r\.platformio\penv` folders 
3. Install the latest Python 2.7.14

Start IDE. All should work.

chall3ng3r, 2017-11-16:
Thank you!

Did as you told, now VSCode starts quickly and PIO extension also loads up without getting stuck on "Verifying". 

Closing this issue.

closed 2017-11-16 by chall3ng3r