platformio/platformio-core #1198: Board support for Macchina M2

John-Titor, 2017-11-22:
It'd be nice to have support for this board:

    https://www.macchina.cc

which is currently supported by the Arduino IDE, and seems to be very similar to the Arduino Due

    http://docs.macchina.cc/m2/technical-references/processor.html

ivankravets, 2017-11-24:
Implemented in https://github.com/platformio/platform-atmelsam/releases/tag/v2.7.1

ivankravets, 2017-11-24:
Would be thankful for a star ⭐️ for PlatformIO Core project https://github.com/platformio/platformio/stargazers

Thanks a lot!

closed 2017-11-24 by ivankravets

NMSTEC, 2021-05-03:
> Would be thankful for a star ⭐️ for PlatformIO Core project https://github.com/platformio/platformio/stargazers
> 
> Thanks a lot!

Unfortunately you guys removed it again in 3.5, is there any chance of returning it into the IDE?
Thanks!