platformio/platformio-core #1199: TI MSP-EXP430FR2433

shaunhey, 2017-11-24:
Please add support for the newest TI MSP430 Launchpad, the TI MSP-EXP430FR2433.  Support for this board was recently added to TI's Arduino fork, Energia.

http://www.ti.com/tool/msp-exp430fr2433

Output of pio device list:
```
/dev/ttyACM1
------------
Hardware ID: USB VID:PID=2047:0013 SER=58349B6F0F000C00 LOCATION=1-1.1.3:1.2
Description: MSP Tools Driver - MSP Application UART

/dev/ttyACM0
------------
Hardware ID: USB VID:PID=2047:0013 SER=58349B6F0F000C00 LOCATION=1-1.1.3:1.0
Description: MSP Tools Driver - MSP Debug Interface
```

ivankravets, 2017-11-24:
Please move an issue to https://github.com/platformio/platform-timsp430/issues

closed 2017-11-24 by ivankravets

shaunhey, 2017-11-25:
Sorry about that, will do!