platformio/platformio-core #1201: Never ending loop ... PlatformIO IDE: Installing... -> Restart ... What to do?

profra, 2017-11-25:
[image]

[image]

What kind of issue is this?

- [X ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [X ] PlatformIO Core. If you’ve found a bug, please provide an information below.
------------------------------------------------------------------
### Configuration
**Operating system**:
1st machine ... Linux Mint
2nd machine ... W10 Pro
**PlatformIO Version** (`platformio --version`):
trying to update from ...b3 to ....b4 ... both machine's-IDEs got stuck

ivankravets, 2017-11-25:
Could you try this solution?

https://github.com/crsmithdev/arrow/issues/495#issuecomment-346939862

profra, 2017-11-25:
Thanks... it helped

closed 2017-11-25 by ivankravets