platformio/platformio-core #1204: 3.5b6: ESPAsyncWebServer library vanished

andig, 2017-11-26:
After upgrading to b6 due to https://github.com/platformio/platformio-core/issues/1173#issuecomment-346961471 this happens:

        .....
	ESPAsyncTCP @ 1.1.3 has been successfully installed!
	Updating ESPAsyncWebServer               @ 1.1.0          [1.1.1]
	Uninstalling ESPAsyncWebServer @ 1.1.0:         [OK]
	Looking for ESPAsyncWebServer library in registry
	Error: Library `ESPAsyncWebServer` has not been found in PlatformIO Registry.
	You can ignore this message, if `ESPAsyncWebServer` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.

	~/htdocs/vzero master*
	❯ pio run -e esp8266
	[Sun Nov 26 21:08:26 2017] Processing esp8266 (build_flags: -Tesp8266.flash.4m1m.ld; upload_port: vzero-edd834.local; lib_deps: ESPAsyncWebServer@^1.1, OneWire@^2.3, DHT Sensor Library@^1.3, DallasTemperature@^3.7, ArduinoJson@^5.1, ESPAsyncTCP@^1.1.2; lib_ignore: AsyncTCP; platform: https://github.com/platformio/platform-espressif8266.git; board: esp12e; framework: arduino; lib_ldf_mode: deep; lib_compat_mode: 1)
	---------------------------------------------------------------------------------------------------------------------------------------
	Warning! Library `ESPAsyncWebServer` has not been found in PlatformIO Registry.
	You can ignore this message, if `ESPAsyncWebServer` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.

ivankravets, 2017-11-26:
The library was renamed to http://platformio.org/lib/show/306/ESP%20Async%20WebServer to be compatible with `library.properties` (Arduino IDE) name.

closed 2017-11-26 by ivankravets

andig, 2017-11-27:
Wouldn‘t it have been better to fix this upstream? I don‘t care about the Arduino Ide at all and this change will probably break a few people‘s builds and makes the naming of the various async libraries again more inconsistent?