platformio/platformio-core #1205: Platformio can't find tool-stm32duino

acrap, 2017-11-27:
When i try to upload firmware (build works fine) i get:

Executing task: platformio.exe run --target upload <

[11/24/17 18:06:45] Processing nucleo_l476rg (platform: ststm32; board: nucleo_l476rg; framework: stm32cube)

PackageManager: Installing tool-stm32duino @ ~1.0.0
Downloading [------------------------------------] 1%
Please manually remove file C:\Users\{USER}\.platformio\packages\_tmp_installing-gc2bho-package\tool-stm32duino-windows-1.0.0.tar.gz
Warning! Package Mirror: [Error 32]

It's not about user privilege policy (i tried to run platformio IDE as admin)

When i run "platformio platform list" i get:
ststm32 ~ ST STM32

The STM32 family of 32-bit Flash MCUs based on the ARM Cortex-M processor is designed to offer new degrees of freedom to MCU users. It offers a 32-bit product range that combines very high performance, real-time capabilities, digital signal processing, and low-power, low-voltage operation, while maintaining full integration and ease of development.

Home: http://platformio.org/platforms/ststm32
Frameworks: arduino, cmsis, libopencm3, mbed, spl, stm32cube
Packages: tool-stm32duino, framework-libopencm3, framework-arduinoststm32, toolchain-gccarmnoneeabi, tool-stlink, framework-arduinostm32mxchip, tool-openocd, framework-spl, framework-stm32cube, framework-cmsis, framework-mbed
Version: 3.6.0

Environment: Win10 x64. VS Code(1.18.1) Platformio Core(3.5.0b3)

I tried to install tool-stm32duino:
platformio platform install tool-stm32duino
PlatformManager: Installing tool-stm32duino
Error: Detected unknown package 'tool-stm32duino'

ivankravets, 2017-11-27:
Please download file from https://bintray.com/platformio/dl-packages/tool-stm32duino#files and unpack manually to `C:\Users/{USER}/.platformio/packages`

closed 2017-11-27 by ivankravets

acrap, 2017-11-29:
It helps. Thank you