platformio/platformio-core #1206: 3.5rc1: ldf_compat_mode=2 breaks non-pio libraries

andig, 2017-11-27:
I'm developing a multi-platform solution for esp8266 and esp32. This requires `ldf_compat_mode=2` for `ESP Async Webserver` and its dependencies.

This breaks the `DHT` arduino library due to platform mismatch:

    ❯ sudo pio upgrade

    Please wait while upgrading PlatformIO ...
    PlatformIO has been successfully upgraded to 3.5.0rc1
    Release notes: http://docs.platformio.org/en/latest/history.html

    ❯ pio run -e esp32    
    Please wait while upgrading PlatformIO...
    PlatformIO has been successfully upgraded to 3.5.0rc1!

    ***************************************************************************************************************************************
    If you like PlatformIO, please:
    - follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
    - star it on GitHub > https://github.com/platformio/platformio
    - try PlatformIO IDE for IoT development > http://platformio.org/platformio-ide
    - support us with PlatformIO Plus > https://pioplus.com
    ***************************************************************************************************************************************

    [Mon Nov 27 10:11:44 2017] Processing esp32 (framework: arduino; lib_deps: ESP Async WebServer@^1.1, OneWire@^2.3, DHT Sensor Library@^1.3, DallasTemperature@^3.7, ArduinoJson@^5.1, https://github.com/bbx10/DNSServer_tng, AsyncTCP@^1.0; lib_ignore: ESPAsyncTCP; platform: https://github.com/platformio/platform-espressif32.git#feature/stage; board: featheresp32; lib_ldf_mode: deep; lib_compat_mode: 2)
    ---------------------------------------------------------------------------------------------------------------------------------------
    PlatformManager: Installing platform-espressif32
    git version 2.13.6 (Apple Git-96)
    Cloning into '/Users/andig/.platformio/platforms/_tmp_installing-DgQzGb-package'...
    remote: Counting objects: 107, done.
    remote: Compressing objects: 100% (89/89), done.
    remote: Total 107 (delta 59), reused 38 (delta 10), pack-reused 0
    Receiving objects: 100% (107/107), 38.40 KiB | 4.27 MiB/s, done.
    Resolving deltas: 100% (59/59), done.
    espressif32 @ 2289af1 has been successfully installed!
    The platform 'https://github.com/platformio/platform-espressif32.git#feature/stage' has been successfully installed!
    The rest of packages will be installed automatically depending on your build environment.
    PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
    Downloading  [####################################]  100%          
    Unpacking  [####################################]  100%          
    PackageManager: Installing framework-arduinoespressif32
    git version 2.13.6 (Apple Git-96)
    Cloning into '/Users/andig/.platformio/packages/_tmp_installing-vPau17-package'...
    remote: Counting objects: 1491, done.
    remote: Compressing objects: 100% (1258/1258), done.
    remote: Total 1491 (delta 198), reused 927 (delta 117), pack-reused 0
    Receiving objects: 100% (1491/1491), 19.55 MiB | 3.74 MiB/s, done.
    Resolving deltas: 100% (198/198), done.
    Submodule 'libraries/BLE' (https://github.com/nkolban/ESP32_BLE_Arduino.git) registered for path 'libraries/BLE'
    Cloning into '/Users/andig/.platformio/packages/_tmp_installing-vPau17-package/libraries/BLE'...
    remote: Counting objects: 216, done.        
    remote: Compressing objects: 100% (35/35), done.        
    remote: Total 216 (delta 3), reused 34 (delta 1), pack-reused 180        
    Receiving objects: 100% (216/216), 167.12 KiB | 566.00 KiB/s, done.
    Resolving deltas: 100% (92/92), done.
    Submodule path 'libraries/BLE': checked out 'f8fe9d7cdfb20caa54b70849826d1ac6e375ff78'
    PackageManager: Installing tool-esptoolpy @ ~1.20100.0
    Downloading  [####################################]  100%          
    Unpacking  [####################################]  100%
    CorePackageManager: Installing tool-scons @ ~3.20501.2
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    tool-scons @ 3.20501.2 has been successfully installed!
    Verbose mode can be enabled via `-v, --verbose` option
    Converting vzero.ino
    Collected 24 compatible libraries
    Looking for dependencies...
    Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
    |-- <ESP Async WebServer> v1.1.1
    |   |-- <AsyncTCP> v1.0.1
    |   |-- <FS> v1.0
    |   |-- <WiFi> v1.0
    |   |-- <ArduinoJson> v5.11.2
    |-- <OneWire> v2.3.2
    |-- <DallasTemperature> v3.7.7
    |   |-- <OneWire> v2.3.2
    |-- <ArduinoJson> v5.11.2
    |-- <DNSServer> v1.1.0
    |   |-- <WiFi> v1.0
    |-- <AsyncTCP> v1.0.1
    |-- <FS> v1.0
    |-- <HTTPClient> v1.1
    |   |-- <WiFiClientSecure> v1.0
    |   |   |-- <WiFi> v1.0
    |   |-- <WiFi> v1.0
    |-- <WiFi> v1.0
    |-- <SPIFFS> v1.0
    |   |-- <FS> v1.0
    |-- <ArduinoOTA> v1.0
    |   |-- <ESPmDNS> v1.0
    |   |-- <WiFi> v1.0
    |   |-- <Update> v1.0
    |-- <ESPmDNS> v1.0
    Compiling .pioenvs/esp32/src/config.o
    Compiling .pioenvs/esp32/src/plugins/AnalogPlugin.o
    Compiling .pioenvs/esp32/src/plugins/DHTPlugin.o
    Compiling .pioenvs/esp32/src/plugins/OneWirePlugin.o
    In file included from src/plugins/OneWirePlugin.h:4:0,
    from src/config.cpp:23:
    .piolibdeps/OneWire_ID1/OneWire.h:245:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
    #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
    ^
    In file included from src/config.cpp:27:0:
    src/plugins/DHTPlugin.h:4:17: fatal error: DHT.h: No such file or directory
    compilation terminated.
    *** [.pioenvs/esp32/src/config.o] Error 1
    In file included from src/plugins/OneWirePlugin.h:4:0,
    from src/plugins/OneWirePlugin.cpp:2:
    .piolibdeps/OneWire_ID1/OneWire.h:245:2: warning: #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture." [-Wcpp]
    #warning "OneWire. Fallback mode. Using API calls for pinMode,digitalRead and digitalWrite. Operation of this library is not guaranteed on this architecture."
    ^
    In file included from src/plugins/DHTPlugin.cpp:2:0:
    src/plugins/DHTPlugin.h:4:17: fatal error: DHT.h: No such file or directory
    compilation terminated.
    *** [.pioenvs/esp32/src/plugins/DHTPlugin.o] Error 1
    ===================================================== [ERROR] Took 52.21 seconds =====================================================

    ============================================================== [SUMMARY] ==============================================================
    Environment esp8266     [SKIP]
    Environment esp32       [ERROR]
    ===================================================== [ERROR] Took 52.21 seconds =====================================================

pio file:

    [platformio]
    #env_default=esp8266
    #env_default=esp32

    [common_env_data]
    lib_deps=
      ESP Async WebServer@^1.1
    # Use git version for _stage compatibility (https://github.com/me-no-dev/ESPAsyncWebServer/issues/281)
    #  https://github.com/me-no-dev/ESPAsyncWebServer.git
      OneWire@^2.3
      DHT Sensor Library@^1.3
      DallasTemperature@^3.7
      ArduinoJson@^5.1

    [env:esp8266]
    #platform=espressif8266
    # use stage version for lwip2 support
    platform=https://github.com/platformio/platform-espressif8266.git#feature/stage
    board=esp12e
    framework=arduino
    # lib_compat_mode=1 allows non-git versions of DHT and ESPAsyncWebServer@^1.1
    lib_compat_mode=1
    lib_ldf_mode=deep
    build_flags=-Tesp8266.flash.4m1m.ld
    upload_port=vzero-edd834.local
    #targets=upload
    lib_deps=
      ${common_env_data.lib_deps}
      ESPAsyncTCP@^1.1.2
    lib_ignore=
      AsyncTCP

    [env:esp32]
    # use stage version for added SPIFFS support
    #platform=espressif32
    platform=https://github.com/platformio/platform-espressif32.git#feature/stage
    board=featheresp32
    framework=arduino
    # lib_compat_mode=2 to avoid including ESPAsyncTCP
    lib_compat_mode=2
    lib_ldf_mode=deep
    lib_deps=
      ${common_env_data.lib_deps}
      https://github.com/bbx10/DNSServer_tng
      AsyncTCP@^1.0
    #  https://github.com/me-no-dev/AsyncTCP.git
    lib_ignore=
      ESPAsyncTCP

ivankravets, 2017-11-27:
http://platformio.org/lib/show/19/DHT%20sensor%20library

Comparison is case sensitive

closed 2017-11-27 by ivankravets

andig, 2017-11-27:
Quite confusing since the library did install just fine. The lib download does seme to apply different (or inconsistent) rules.

After cleaning the `.pio*` folders the error has changed:

	Archiving .pioenvs/esp32/lib/libOneWire_ID1.a
	Indexing .pioenvs/esp32/lib/libOneWire_ID1.a
	Compiling .pioenvs/esp32/lib/DHT sensor library_ID19/DHT.o
	Compiling .pioenvs/esp32/lib/DHT sensor library_ID19/DHT_U.o
	In file included from .piolibdeps/DHT sensor library_ID19/DHT_U.cpp:22:0:
	.piolibdeps/DHT sensor library_ID19/DHT_U.h:25:29: fatal error: Adafruit_Sensor.h: No such file or directory
	#include <Adafruit_Sensor.h>
	^
	compilation terminated.
	*** [.pioenvs/esp32/lib/DHT sensor library_ID19/DHT_U.o] Error 1

It seems nested includes need manually be installed? This was working before:

	❯ ls .piolibdeps 
	ArduinoJson_ID64          DHT sensor library_ID19   DallasTemperature_ID54    ESPAsyncTCP_ID305
	AsyncTCP_ID1826           DNSServer                 ESP Async WebServer_ID306 OneWire_ID1

Now it needs `Adafruit Unified Sensor` in addition in the pio file?

ivankravets, 2017-11-27:
We switched to Adafruit official library manifest which doesn’t contain dependencies information. They use library.properties, old-school Arduino IDE manifest

You have to specify extra dependencies in project config

andig, 2017-11-27:
Good to know, thank you. Compiling fine now.