platformio/platformio-core #1208: Home: Could not load recent projects

robinharris, 2017-11-28:
Server error

ivankravets, 2017-11-28:
Could you reproduce this issue after a restart of IDE?

robinharris, 2017-11-28:
Yes.  I restarted the IDE several times and the problem persisted.  Then I rebooted my MacBook Pro and the problem still persisted.  Finally I deleted all the project folders and from then on it worked normally.

Robin

> 
> Could you reproduce this issue after a restart of IDE?

ivankravets, 2017-11-28:
Next time when you will have the same issue, please open new and attach `~/.platformio/homestate.json` here. It will help us to debug.

closed 2017-11-28 by ivankravets

robinharris, 2017-11-28:
OK, will do that. Thanks

Robin

> 
> Next time when you will have the same issue, please open new and attach ~/.platformio/homestate.json here. It will help us to debug.