platformio/platformio-core #1209: Errors after running pio update

chessweb01, 2017-11-29:
I just ran `pio update` in Visual Studio Code with platformio. Now I get the following error when I open the IDE:

PlatformIO: IntelliSense Index failed: Error: Detected unknown package 'espressif8266_stage'

I have verified that espressif8266_stage is installed correctly by running `platformio platform install https://github.com/platformio/platform-espressif8266.git#feature/stage`, so that is not the problem.

When I try to build a project using the espressif8266_stage platform I get 

PlatformManager: Installing espressif8266_stage
Error: Detected unknown package 'espressif8266_stage'
The terminal process terminated with exit code: 1

in the console. What is going on?

ivankravets, 2017-11-29:
Please update http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

closed 2017-11-29 by ivankravets

me21, 2017-12-01:
Try to use `platform = espressif8266@4f59cef` or any other version that is output by `pio platform list` command.

ivankravets, 2017-12-01:
PIO Core 3.5 + http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

ivankravets, 2017-12-01:
@me21 PIO doesn't update dev/platforms or libraries by default. However, you can change it http://docs.platformio.org/en/latest/userguide/cmd_settings.html#auto-update-platforms

sarfata, 2017-12-29:
@ivankravets Sorry to bug you again with this issue but I also do not understand this.

I have:
```
[env:esp]
platform=https://github.com/platformio/platform-espressif8266.git#feature/stage
framework = arduino
```

And when I try to build, I get:
```
espressif8266 @ 572bcc6 has been successfully installed!
The platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
Error: Unknown development platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage'
```

I can make it work, if I change to:
```
platform=espressif8266@572bcc6
```

But this is **not** what you have in that page that you keep linking to: http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version. 

It also sucks that users have to manually run `platformio platform install https://github.com/platformio/platform-espressif8266.git#572bcc6` 😞 

Can you clarify what we are supposed to do? And I think the doc should be updated because right now what the doc says does not work.

Related issues: 
 - https://github.com/gepd/Deviot/issues/179
 - https://github.com/platformio/platform-espressif8266/issues/63
 - https://community.platformio.org/t/detected-unknown-package-espressif8266-stage/981/12

me21, 2017-12-29:
Hmmm, it worked for me when I did as the docs say. What Platformio Core version do you have?