platformio/platformio-core #1214: Home: Internal Store Exception

ajitj, 2017-12-04:
TypeError: Cannot read property 'name' of null
    at http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:11:33138
    at Array.map (native)
    at _ (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:11:33009)
    at http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:11:33528
    at Array.map (native)
    at y (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:11:33468)
    at Function.a [as mapToProps] (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:31:328504)
    at r (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:6:10560)
    at l (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:6:11948)
    at c (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:6:12066)

ivankravets, 2017-12-04:
Could you explain how to reproduce this issue? Do you open any library?

ajitj, 2017-12-05:
I think platformio ide extension setup in visual studio didn't complete due to network issue. With a good network connection when I uninstalled (along with dependencies) and installed the extension it started working correctly.

Thanks for the response. We can close this

closed 2017-12-05 by ajitj