platformio/platformio-core #1215: Home: Internal Store Exception

SashaShcherbinin, 2017-12-05:
TypeError: Cannot read property 'name' of null
    at http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:11:33138
    at Array.map (native)
    at _ (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:11:33009)
    at http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:11:33528
    at Array.map (native)
    at y (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:11:33468)
    at Function.a [as mapToProps] (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:31:328504)
    at r (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:6:10560)
    at l (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:6:11948)
    at c (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:6:12066)

ivankravets, 2017-12-05:
Could you explain how to reproduce this issue?

ivankravets, 2017-12-13:
Duplicate of https://github.com/platformio/platformio-core/issues/1225

closed 2017-12-13 by ivankravets