platformio/platformio-core #1222: Home: Error occured while requesting content from http://dl.platformio.org/libraries/examples/2/291/ClickEncoderTest.ino

davidbitton, 2017-12-12:
Error: Request has been terminated
Possible causes: the network is offline, Origin is not allowed by Access-Control-Allow-Origin, the page is being unloaded, etc.
    at u.crossDomainError (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:25:174668)
    at XMLHttpRequest.u._end.t.onreadystatechange (http://127.0.0.1:8010/e260ca4c36f1307bfff8.min.js:25:175875)

ivankravets, 2017-12-12:
Could you reproduce this issue after IDE restart?

davidbitton, 2017-12-13:
Yes. I think it has to do with our firewall at work. It mangles requests
with bad cross site headers. However, the page comes up if I come back to
the library page after install.

ivankravets, 2017-12-13:
Thanks for the report!

closed 2017-12-13 by ivankravets