platformio/platformio-core #1223: Home: Could not update platform

guettomusick, 2017-12-12:
PIO Core Call Error: Platform /Users/fer/.platformio/platforms/wiznet7500
--------
Updating wiznet7500                      @ 15622ec        [e91f049]
git version 2.8.0

fatal: Not a git repository: /Users/fer/.platformio/platforms/_tmp_installing-fTG_Y5-package/.git/modules/builder/frameworks/mbed
Error: VCS: Could not process command ['git', 'pull', '--recurse-submodules']

ivankravets, 2017-12-12:
1. Do you see this folder on a disk? `/Users/fer/.platformio/platforms/_tmp_installing-fTG_Y5-package`
2. Please provide a content of `/Users/fer/.platformio/platforms/wiznet7500/.git/.piopkgmanager.json`

jfessard, 2017-12-14:
I am seeing a very similar error message when updating the platform since pio asked me to, and the tempfolder in the error message (lwip2) doesn't exist.
```
bench@bench ~/src/espbtc $ pio update
Updating tool-scons                      @ 3.20501.2      [Up-to-date]
Updating tool-unity                      @ 1.20302.1      [Up-to-date]
Updating pysite-pioplus                  @ 0.4.2          [Up-to-date]
Updating tool-pioplus                    @ 0.11.0         [Up-to-date]

Platform Manager
================
Platform Espressif 8266 (Stage)
--------
Updating espressif8266                   @ 4f59cef        [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ eb49406        [58937dd]
git version 2.7.4
fatal: Not a git repository: /home/bench/.platformio/packages/_tmp_installing-HzqBQ8-package/.git/modules/lwip2
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 104, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/update.py", line 46, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/platform.py", line 376, in platform_update
    platform, only_packages=only_packages, only_check=only_check)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/platform.py", line 123, in update
    p.update_packages(only_check)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/platform.py", line 285, in update_packages
    self.pm.update(manifest['__pkg_dir'], version, only_check)
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/package.py", line 731, in update
    assert vcs.update()
  File "/usr/local/lib/python2.7/dist-packages/platformio/vcsclient.py", line 156, in update
    return self.run_cmd(args)
  File "/usr/local/lib/python2.7/dist-packages/platformio/vcsclient.py", line 96, in run_cmd
    return check_call(args, **kwargs) == 0
  File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
CalledProcessError: Command '['git', 'pull', '--recurse-submodules']' returned non-zero exit status 1
```
Contents of my `.piopkgmanager.json `:

`{"url": "git+https://github.com/platformio/platform-espressif8266.git#feature/stage", "version": "4f59cef", "requirements": null, "name": "platform-espressif8266"}`

Manually uninstalling the platform and reinstalling the staging one again seemed to have worked.

ivankravets, 2017-12-14:
@jfessard it seems that bug with Git. Do you see any tracks with "_tmp_installing" in `/home/bench/.platformio/packages`? You can do the next:
```
ag _tmp_installing /home/bench/.platformio/packages
```

jfessard, 2017-12-16:
I didn't find any _tmp_installing files in there, but since I already manually removed and reinstalled the staging esp8266 it would explain it.
Sadly I didn't keep it in the "failed" state since I had to work on it.

ivankravets, 2017-12-16:
Please reopen this issue next time when you can reproduce it. Thanks!

ivankravets, 2017-12-23:
My git version is 2.14.3, I can't reproduce this issue. Could upgrade your Git client and re-install dev/platform?

jfessard, 2018-01-05:
OK so I updated git (with the proper ppa) and platformio since there's a new version. Now it's stuck, I can't even uninstall the platform to reinstall the new one (I might have to remove the platform manually)
Whatever `pio` command I run, it tries to upgrade platformio and fails on the framework.
```
bench@bench ~/src/ESProjector $ pio run
Please wait while upgrading PlatformIO...
Platform Espressif 8266 (Stage)
--------
Updating espressif8266                   @ 3352890        [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.409.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ d0a4900        [89837fc]
git version 2.15.1
fatal: Not a git repository: /home/bench/.platformio/packages/_tmp_installing-xHkttl-package/.git/modules/lwip2
Error: VCS: Could not process command ['git', 'pull', '--recurse-submodules']
bench@bench ~/src/ESProjector $ pio --version
PlatformIO, version 3.5.0
bench@bench ~/src/ESProjector $ 
```

ivankravets, 2018-01-08:
Please reopen if you will have this issue in future

ivankravets, 2019-06-01:
Should be fixed in the latest release.

closed 2019-06-01 by ivankravets

TD-er, 2019-09-18:
I had the same issue and could also work around it by removing the stated directories.
But since that is not really a proper fix, I was wondering if it may be related to some unknown tag or branch in some of the sub modules of the branch?

As a reference my latest output of it given in a Travis build:
```
20.78s$ platformio update
Updating tool-scons                      @ 2.20501.190826 [Up-to-date]
Platform Manager
================
Platform Espressif 32
--------
Updating espressif32                     @ 1.9.0          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 2.10002.190628 [Up-to-date]
Updating tool-openocd-esp32              @ 1.1000.20190708[Up-to-date]
Platform Espressif 8266
--------
Updating espressif8266                   @ 2.2.3          [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20502.0      [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating toolchain-xtensa                @ 2.40802.190218 [Up-to-date]
Platform Espressif 8266
--------
Updating espressif8266                   @ 7507605        [Up-to-date]
Updating framework-arduinoespressif8266  @ 1.20401.3-puya [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
Platform Espressif 8266 (Stage)
--------
Updating espressif8266                   @ 4db7ada        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating framework-arduinoespressif8266  @ 7ee793e        [645464c]
git version 2.23.0
remote: Enumerating objects: 22846, done.
remote: Coremote: Counting objects: 100% (22844/22844), done.
remote: Compressing objects: 100% (6994/6994), done.
remote: Total 21835 (delta 14324), reused 20879 (delta 13372), pack-reused 0
Receiving objects: 100% (21835/21835), 57.89 MiB | 14.00 MiB/s, done.
Resolving deltas: 100% (14324/14324), completed with 700 local objects.
From https://github.com/Jason2866/Arduino
 + 7ee793e7...645464ca pre_26x    -> origin/pre_26x  (forced update)
 * [new tag]           2.0.0       -> 2.0.0
 * [new tag]           2.0.0-rc1   -> 2.0.0-rc1
 * [new tag]           2.0.0-rc2   -> 2.0.0-rc2
 * [new tag]           2.1.0       -> 2.1.0
 * [new tag]           2.1.0-rc1   -> 2.1.0-rc1
 * [new tag]           2.1.0-rc2   -> 2.1.0-rc2
 * [new tag]           2.2.0       -> 2.2.0
 * [new tag]           2.2.0-rc1   -> 2.2.0-rc1
 * [new tag]           2.3.0       -> 2.3.0
 * [new tag]           2.3.0-rc1   -> 2.3.0-rc1
 * [new tag]           2.3.0-rc2   -> 2.3.0-rc2
 * [new tag]           2.4.0       -> 2.4.0
 * [new tag]           2.4.0-rc1   -> 2.4.0-rc1
 * [new tag]           2.4.0-rc2   -> 2.4.0-rc2
 * [new tag]           2.4.1       -> 2.4.1
 * [new tag]           2.4.2       -> 2.4.2
 * [new tag]           2.5.0       -> 2.5.0
 * [new tag]           2.5.0-beta1 -> 2.5.0-beta1
 * [new tag]           2.5.0-beta2 -> 2.5.0-beta2
 * [new tag]           2.5.0-beta3 -> 2.5.0-beta3
 * [new tag]           2.5.1       -> 2.5.1
 * [new tag]           2.5.2       -> 2.5.2
Fetching submodule libraries/ESP8266SdFat
Fetching submodule libraries/LittleFS/lib/littlefs
Fetching submodule libraries/SoftwareSerial
From https://github.com/plerup/espsoftwareserial
 + dd9d232...776d49b master     -> origin/master  (forced update)
 * [new tag]         5.3.0      -> 5.3.0
Fetching submodule tools/esptool
Fetching submodule tools/pyserial
Fetching submodule tools/sdk/lwip2/builder
Fetching submodule tools/sdk/lwip2/builder/lwip2-src
Fetching submodule tools/sdk/ssl/bearssl
From https://github.com/earlephilhower/bearssl-esp8266
 * branch            0d7d5e2e59c240703ac64e17177b044df5b3748a -> FETCH_HEAD
From https://github.com/plerup/espsoftwareserial
 * branch            b178c0ff023ff87a4754636b624a6d18e6036a07 -> FETCH_HEAD
fatal: refusing to merge unrelated histories
Error: VCS: Could not process command ['git', 'pull', '--recurse-submodules']
The command "platformio update" exited with 1.
```
This does look a bit like git does not know what SHA to use.

Could it be fixed by running a `git fetch` on these sub modules?
If so, then it is still the question whether this is a PIO issue, or a Git issue.

ivankravets, 2019-09-18:
Did you change files directly in `framework-arduinoespressif8266`?

TD-er, 2019-09-18:
> Did you change files directly in `framework-arduinoespressif8266`?

Nope, but now I come to think of it, one of the repos I use is not directly the esp8266/Arduino repo, but one maintained by @Jason2866 https://github.com/Jason2866/platform-espressif8266.git#Tasmota

I will switch that one to the normal staged one and see if that fixes it in Travis (which is still giving the error)

TD-er, 2019-09-18:
With the platform set to the original stage branch, it also returns the same error when updating it.
See 
- [Travis log with platformio defined platform](https://travis-ci.org/letscontrolit/ESPEasy/builds/586476612?utm_source=github_status&utm_medium=notification)
- [Travis log with Jason2866 defined platform](https://travis-ci.org/letscontrolit/ESPEasy/builds/586467491?utm_source=github_status&utm_medium=notification)

Jason2866, 2019-09-18:
@TD-er  Was a error in Arduino repo fixed with https://github.com/esp8266/Arduino/pull/6526
I updated my Arduino fork which is called from https://github.com/Jason2866/platform-espressif8266.git#Tasmota

Update all pio stuff

Jason2866, 2019-09-18:
Do you use caching for Travis? 
We disabled that completly for Tasmota (did make weird issues)

TD-er, 2019-09-18:
Should I disable all caches or just not include the 
`.platformio` dir and `.pio` dir?

Jason2866, 2019-09-18:
We disabled all. Made no further investigations since disabling all doesnt changed (for us) Travis build time