platformio/platformio-core #1227: Linkit Smart Duo (ARDUINO MCU PROGRAMING) Support Request

ssribeiro, 2017-12-13:
What kind of issue is this?

- [ ] Feature Request. 

*You can erase any parts of this template not applicable to your Issue.*
------------------------------------------------------------------

### Additional info

Can add support for Linkit Smart Duo platform from MediaTek for programing in arduino language?
It would be nice of you.
Thanks.

ivankravets, 2017-12-13:
Added https://github.com/platformio/platform-atmelavr/commit/0f48c2b69bf57a36be9d26346861da6e26192cf8

ivankravets, 2017-12-13:
Please test it with this `platformio.ini`
```ini
[env:smart7688]
platform = https://github.com/platformio/platform-atmelavr.git
framework = arduino
board = smart7688
```

Does it work?

closed 2017-12-13 by ivankravets