platformio/platformio-core #1234: No stuch commande on : everything

Essenceia, 2017-12-15:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

Python 2.7.14

**Operating system**:

4.9.66-1-MANJARO (arch)

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 3.4.1
+ have  99-platformio-udev.rules 

### Description of problem

No more commands seem to be recognied, even `platformio init` and `platformio init` isn't recognised ....
`Error: No such command "init"` 

Only `platformio --version` still seems to work. I have already unistalled + reinstalled with pip as `pip uninstall platformio` `pip install -U platformio` with no results.

#### Steps to Reproduce

1.Install serial module `pip install serial`
2.Uninstall serial module `pip unistall serial`

### Additional info

If field is left empty `platformio` I get `Error: No such command "ci"`.
Thanks for the help, I have no idea where to start with this problem.

ivankravets, 2017-12-16:
You need to use virtual environment in this case if main `python` binary is 3.0. We don't support Python 3.0 yet.

```
sudo pip install virtualenv
virtualenv ~/.platformio/penv -p pyton2.7
source ~/.platformio/penv/activate
(penv) pip install platformio
```

Now, please add `~/.platformio/penv/bin` to your system `PATH`.

Essenceia, 2017-12-16:
Hello, unfortunately it would seem that the project still persists with the same symptoms. There were no problems with any of the installs.
` echo $PATH `
`/home/pookie/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/home/pookie/.platformio/penv/bin `

ivankravets, 2017-12-16:
> source ~/.platformio/penv/bin/activate

Ahhh... That was my mistake. Sorry :(

I recommend to switch to the latest PIO Core 3.5.0RC:

```
~/.platformio/penv/bin/pip install -U https://github.com/platformio/platformio-core/archive/develop.zip
```

closed 2017-12-16 by ivankravets