platformio/platformio-core #1235: After update to Core 3.5.0rc9 include files not found anymore

beegee-tokyo, 2017-12-16:
### Configuration

**Operating system**: Windows 7 Ultimate 64bit Service Pack 1

**PlatformIO Version** (`platformio --version`): 3.5.0rc9

**Visual Studio Code Version**: 1.18.0

**Atom Version**: 1.23.1 x64
### Description of problem
Include files not found anymore after update to Core 3.5.0rc9

#### Steps to Reproduce

1. Update to latest core 
2. Build existing code which compiled before core update

### Actual Results
```
> Executing task in folder ESP32-Weatherstation: platformio.exe run <

[12/16/17 14:25:46] Processing esp32dev (build_flags: -Os -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG; platform: https://github.com/platformio/platform-espressif32.git#feature/stage; board: esp32dev; framework: arduino; lib_extra_dirs: B:\Projects\ESP32\ESP32Lib; upload_port: ESP32-Test-2481CE9C.local)
----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 37 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <SPI> v1.0
|-- <DHT sensor library for ESP32> v1.0.2
|-- <Adafruit Unified Sensor> v1.0.2
|-- <Adafruit TSL2561> v1.0.0
|   |-- <Adafruit Unified Sensor> v1.0.2
|   |-- <Wire> v1.0
Compiling .pioenvs\esp32dev\src\emq.o
Compiling .pioenvs\esp32dev\src\graphics.o
Compiling .pioenvs\esp32dev\src\leds.o
Compiling .pioenvs\esp32dev\src\light.o
In file included from src\emq.cpp:1:0:
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
In file included from src\leds.cpp:1:0:
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
In file included from src\graphics.cpp:1:0:
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
Compiling .pioenvs\esp32dev\src\main.o
In file included from src\light.cpp:1:0:
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
Compiling .pioenvs\esp32dev\src\ota.o
In file included from src\main.cpp:1:0:
src/setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
Compiling .pioenvs\esp32dev\src\reset_reason.o
In file included from src\ota.cpp:1:0:
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
*** [.pioenvs\esp32dev\src\emq.o] Error 1
C*ompiling .pioenvs\esp32dev\src\setup.o** [.pioenvs\esp32dev\src\leds.o] Error 1

*** [.pioenvs\esp32dev\src\graphics.o] Error 1
*** [.pioenvs\esp32dev\src\light.o] Error 1
*** [.pioenvs\esp32dev\src\main.o] Error 1
*** [.pioenvs\esp32dev\src\ota.o] Error 1
In file included from src\reset_reason.cpp:1:0:
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
*** [.pioenvs\esp32dev\src\reset_reason.o] Error 1
In file included from src\setup.cpp:1:0:
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
*** [.pioenvs\esp32dev\src\setup.o] Error 1
========================================================= [ERROR] Took 1.36 seconds =========================================================
The terminal process terminated with exit code: 1
```

### Expected Results
Under previous Core version the code compiled without any warnings or errors and is uploaded to the ESP32.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
upload_port = ESP32-Test-2481CE9C.local
build_flags = -Os -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
lib_extra_dirs = B:\Projects\ESP32\ESP32Lib
```

**Source file to reproduce issue:**
```cpp
#ifndef setup_h
#define setup_h

// Standard libraries
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <esp_system.h>
#include <rom/rtc.h>

// Declarations and function definitions
#include <TFT_eSPI.h>

#endif
```

### Additional info
After update of PlatfomIO Core to 3.5.0rc9 the **_`xxx.h No such file or directory`_** error occurs on all my projects (ESP32 or ESP8266).
Error occurs on both Visual Studio Code and Atom.
Platforms, packages and libs are all installed.

I tried to remove all projects from the workspace, close Atom (or VSC), delete **`C:\Users\beegee\.platformio\homestate.json`**, restart the editor. Still same problem.

I tried to uninstall PlatformIO from both editors, deleted **`C:\Users\beegee\.platformio`**, reinstalled PlatformIO on both editors. Still same problem.

Don't know how to go back to previous version of core.

beegee-tokyo, 2017-12-16:
### Additional info
**_`platformio run -v --target upload`_** output:
```
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

B:\Projects\ESP32\ESP32-Weatherstation>platformio run -v --target upload
[12/16/17 15:09:29] Processing esp32dev (build_flags: -Os -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG; platform: https://github.com/platformio/platform-espressif32.git#feature/stage; board: esp32dev; framework: arduino; lib_extra_dirs: B:\Projects\ESP32\ESP32Lib; upload_port: ESP32-Test-2481CE9C.local)
----------------------------------------------------------------------------------------------------------------------------------------------
Collected 37 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <Adafruit Unified Sensor> v1.0.2 (C:\Users\beegee\.platformio\lib\Adafruit Unified Sensor_ID31)
|-- <SPI> v1.0 (C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\libraries\SPI)
|-- <DHT sensor library for ESP32> v1.0.2 (B:\Projects\ESP32\ESP32Lib\DHTesp)
|-- <Adafruit TSL2561> v1.0.0 (C:\Users\beegee\.platformio\lib\Adafruit TSL2561_ID34)
|   |-- <Adafruit Unified Sensor> v1.0.2 (C:\Users\beegee\.platformio\lib\Adafruit Unified Sensor_ID31)
|   |-- <Wire> v1.0 (C:\Users\beegee\.platformio\packages\framework-arduinoespressif32\libraries\Wire)
xtensa-esp32-elf-g++ -o .pioenvs\esp32dev\src\emq.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -Os -g3 -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Os -DARDUINO=10805 -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"Espressif ESP32 Dev Module\"" -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DPLATFORMIO=30500 -DARDUINO_ESP32_DEV -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\config 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bluedroid 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\app_trace 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\app_update 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bootloader_support 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bt 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\driver 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp32 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\esp_adc_cal 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\ethernet 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\fatfs 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\freertos 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\heap 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\jsmn 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\log 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mdns 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mbedtls 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\mbedtls_port 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\nvs_flash 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\openssl 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\spi_flash 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\sdmmc 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\spiffs 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\tcpip_adapter 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\ulp 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\vfs 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\wear_levelling 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\xtensa-debug-module 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\console 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\soc 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\newlib 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\coap 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\wpa_supplicant 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\expat 
-IC:\users\beegee\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\json 
[... cut ...]
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
*** [.pioenvs\esp32dev\src\reset_reason.o] Error 1
In file included from src\setup.cpp:1:0:
src\setup.h:22:18: fatal error: WiFi.h: No such file or directory
compilation terminated.
*** [.pioenvs\esp32dev\src\setup.o] Error 1
========================================================= [ERROR] Took 1.27 seconds =========================================================

B:\Projects\ESP32\ESP32-Weatherstation>
```

beegee-tokyo, 2017-12-16:
Another finding. If I change in **`setup.h`** from
**`#include "WiFI.h"`**
to
**`#include "C:/Users/beegee/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFI.h"`**
it compiles. ?????

ivankravets, 2017-12-16:
Please run in PIO IDE Terminal `pio upgrade --dev`, should work now.

Thanks a lot for the report! I fixed a lot of issues with the last commit.

closed 2017-12-16 by ivankravets

beegee-tokyo, 2017-12-16:
Thanks for Saturday before Christmas fix! I'll try tomorrow.
Merry Xmas or happy holidays, whatever you prefer.

commit 3d8183a2 (2017-12-16): LDF: handle "include" folder per project; fix issue with header files // Resolve #1235