platformio/platformio-core #1236: Home: Internal Store Exception

siddhc, 2017-12-16:
TypeError: Cannot read property 'length' of null
    at _ (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:11:33158)
    at http://127.0.0.1:8010/f00d19876293fb32c015.min.js:11:33885
    at Array.map (native)
    at y (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:11:33779)
    at Function.a [as mapToProps] (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:25:403989)
    at r (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:10578)
    at l (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:11948)
    at c (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:12066)
    at http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:12219
    at Object.run (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:6152)

ivankravets, 2017-12-16:
Could you explain how to reproduce this issue?

ivankravets, 2017-12-16:
@tchisaka @RainWithYou @siddhc how to reproduce this issue?

ivankravets, 2017-12-16:
Please open PIO IDE Terminal and type `pio update`. Restart IDE.

closed 2017-12-16 by ivankravets