platformio/platformio-core #1243: Home: Internal Store Exception on PlatformIO(Atom)

adeltc, 2017-12-16:
I can't create a new arduino project. When I click on New project, the Board section is stuck searching with no results and the Framework section is emptly.

I'm using Atom on windows 10 64bit

I keep getting those errors :

TypeError: Cannot read property 'length' of null
    at _ (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:11:33158)
    at http://127.0.0.1:8010/f00d19876293fb32c015.min.js:11:33885
    at Array.map (native)
    at y (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:11:33779)
    at Function.a [as mapToProps] (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:25:403989)
    at r (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:10578)
    at l (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:11948)
    at c (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:12066)
    at http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:12219
    at Object.run (http://127.0.0.1:8010/f00d19876293fb32c015.min.js:6:6152)

ivankravets, 2017-12-16:
Duplicate of https://github.com/platformio/platformio-core/issues/1236

Please open PIO IDE Terminal and type `pio update`. Restart IDE.

closed 2017-12-16 by ivankravets

adeltc, 2017-12-16:
I just did open PIO IDE Terminal and type pio update. Restart IDE and still not working for me

adeltc, 2017-12-16:
Oups, sorry, I did not wait enouph time :) I seems to work now. Thanks!!!!

ivankravets, 2017-12-16:
Sorry for the issue, we fixed it quickly.