platformio/platformio-core #1250: T89C51CC01

heikohettler, 2017-12-19:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

heikohettler, 2017-12-19:
I need this board to start to program

ivankravets, 2017-12-19:
Duplicate of https://github.com/platformio/platformio-core/issues/907

closed 2017-12-19 by ivankravets

chrizbee, 2022-05-13:
> Duplicate of #907

This is marked as `done` here https://registry.platformio.org/platforms/platformio/intel_mcs51 but I can't seem to find the Atmel T89C51CC01 MCU there...

ivankravets, 2022-05-13:
Please file a feature request at https://github.com/platformio/platform-intel_mcs51/issues