platformio/platformio-core #1252: Library keeps reinstalling for non-latin paths

me21, 2017-12-20:
### Configuration

**OS**: Windows 7 SP1 x64 Russian

**PlatformIO Version** (`platformio --version`): 3.5.0rc14

### Description of problem
I've created a simple project which uses ESP Async Webserver library. On every build, it keeps reinstalling the library, even though it is already installed.

It complains about some conversion error:
```
C:\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py:1303: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as be
ing unequal
val = [x for x in val if x not in dk]
```

Moving the project to the path which doesn't contain cyrillic letters fixed it for me.

#### Steps to Reproduce

1. Create the project with non-latin folder path.
2. Build it.
3. Notice the warning (it is present even when no libraries are mentioned in platformio.ini).

ivankravets, 2017-12-28:
A library name is case sensitive. Please use `ESP Async WebServer`

ivankravets, 2017-12-28:
TODO: Replace [manifest['__pkg_dir'] = util.path_to_unicode(pkg_dir)](https://github.com/platformio/platformio-core/blob/0516cd74e121215a46cd05e5306a401791aa9f69/platformio/managers/package.py#L288) with `manifest['__pkg_dir'] = pkg_dir`.

me21, 2018-05-26:
@ivankravets Sometimes library manager even fails to reinstall the library. As a consequence, the build fails.

> LibraryManager: Installing id=136
> Error: Traceback (most recent call last):
>   File "c:\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
>     cli(None, None, None)
>   File "c:\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
>     return self.main(*args, **kwargs)
>   File "c:\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
>     rv = self.invoke(ctx)
>   File "c:\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "c:\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "c:\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
>     return callback(*args, **kwargs)
>   File "c:\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
>     return f(get_current_context(), *args, **kwargs)
>   File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 108, in cli
>     result = (envname, ep.process())
>   File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 207, in process
>     result = self._run()
>   File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 302, in _run
>     self.verbose)
>   File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 325, in _autoinstall_libdeps
>     ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
>   File "c:\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
>     return callback(*args, **kwargs)
>   File "c:\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
>     return f(get_current_context().obj, *args, **kwargs)
>   File "c:\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 103, in lib_install
>     library, silent=silent, interactive=interactive, force=force)
>   File "c:\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 354, in install
>     force=force)
>   File "c:\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 694, in install
>     pkg_dir = self._install_from_piorepo(name, requirements)
>   File "c:\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 213, in _install_from_piorepo
>     dl_data['url'], requirements)
>   File "c:\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 421, in _install_from_url
>     dlpath = self.download(url, tmp_dir, sha1)
>   File "c:\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 176, in download
>     dst_path = join(dest_dir, fname)
>   File "c:\.platformio\penv\lib\ntpath.py", line 85, in join
>     result_path = result_path + p_path
> UnicodeDecodeError: 'ascii' codec can't decode byte 0xc7 in position 7: ordinal not in range(128)
> 
> ============================================================
> 
> An unexpected error occurred. Further steps:
> 
> * Verify that you have the latest version of PlatformIO using
>   `pip install -U platformio` command
> 
> * Try to find answer in FAQ Troubleshooting section
>   http://docs.platformio.org/page/faq.html
> 
> * Report this problem to the developers
>   https://github.com/platformio/platformio-core/issues
> 
> ============================================================

ivankravets, 2018-05-26:
See temporary solution http://docs.platformio.org/en/latest/faq.html#unicodedecodeerror-non-ascii-characters-found-in-build-environment

You can set [PLATFORMIO_HOME_DIR](http://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_HOME_DIR) environment variable to `C:\.platformio` and avoid non-ascii chars in project path.

I'll look into this issue soon.

ivankravets, 2019-06-05:
Please open PlatformIO IDE Terminal and type
```
pio upgrade --dev
pio update
```

Restart VSCode. Does it work now?

commit 84ce7db3 (2019-06-05): Fixed an issue when library keeps reinstalling for non-latin path // Resolve #1252
closed 2019-06-05 by ivankravets