platformio/platformio-core #1254: No ARMv8 host support (libpython2.7.so.1.0 error)

ristomatti, 2017-12-20:
### Configuration

**Operating system**: 
Linux odr01 3.14.79-113 #1 SMP PREEMPT Wed May 10 00:43:15 BRT 2017 aarch64 aarch64 aarch64 GNU/Linux

**PlatformIO Version**:
PlatformIO, version 3.4.1

### Description of problem
PlatformIO `pioplus` does not support ARMv8 hosts. I have installed PlatformIO on an Odroid C2 with ´pip install -U --user platformio` but the CLI commands do not seem to work:

```
$ pio remote device list
/home/odroid/.platformio/packages/tool-pioplus/pioplus: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory
```

libpython2.7.so.1.0 is located at /usr/lib/aarch64-linux-gnu/libpython2.7.so.1.0:
```
$ find /usr/lib/ -name libpython2.7.so.1.0
/usr/lib/aarch64-linux-gnu/libpython2.7.so.1.0
```
If LD_LIBRARY_PATH is given I get wrong ELF class error:
```
$ LD_LIBRARY_PATH=/usr/lib/aarch64-linux-gnu platformio remote device list
/home/odroid/.platformio/packages/tool-pioplus/pioplus: error while loading shared libraries: libpython2.7.so.1.0: wrong ELF class: ELFCLASS64
```
I have an Raspberry Pi 3 and on it got no errors. ARMv8 should be backwards compatible with ARMv7.

#### Steps to Reproduce

1. Install PlatformIO using `pip` on an ARMv8 host such as the Odroid C2
2. Run a PlatformIO CLI command that requires libpython2.7

### Actual Results
Unable to run CLI commands, but receiving `pioplus/pioplus: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory`

### Expected Results
Command runs succesfully.

ristomatti, 2018-02-20:
Any hope for getting this resolved or something I could do to help resolving it? I've specifically subscribed for the remote agent feature to hook up my development boards to my Odroid C2.

ivankravets, 2018-06-18:
Does someone still use Odroid? I would be glad to provide some patched version of `pioplus` for the testing.

ristomatti, 2018-06-18:
@ivankravets Yes! I still use it and would still like to attach devices to it for remote debugging. Where can I get the patched version for testing?

ivankravets, 2018-06-18:
Please unpack that archive on your Odroid side and navigate to the folder where `test.so` is available. Now run `python -c "import test"`
[test.so.zip](https://github.com/platformio/platformio-core/files/2112264/test.so.zip)

Please share an output here.

ristomatti, 2018-06-18:
This is what I'm getting:

```
$ python -c "import test"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: ./test.so: wrong ELF class: ELFCLASS32
```

ivankravets, 2018-06-18:
> I have an Raspberry Pi 3

Just for kicks, could try https://github.com/platformio/platformio-core/issues/1254#issuecomment-398129390 on the RPi3?

ristomatti, 2018-06-18:
Fair enough. Let me know if I can help with something. I don't have an python experience though.

As a side note I also had a problem with a binary when using fliclib-linux-hci and reported an issue. The developer there managed to create a binary for testing within the same day. I would assume it was cross compiled in some manner as it seems there aren't too many of these ARMv8 devices around. https://github.com/50ButtonsEach/fliclib-linux-hci/issues/62

ivankravets, 2018-06-18:
I mean just repeat the same steps with `test.so` on RPI3. See https://github.com/platformio/platformio-core/issues/1254#issuecomment-398129390

ristomatti, 2018-06-18:
On RPi 3 I get the below output, which I assume is the one you thought could work on the Odroid.

```
$ python -c "import test"
sys.version_info(major=2, minor=7, micro=9, releaselevel='final', serial=0)
<module 'platformio.commands.platform' from '/home/pi/.local/lib/python2.7/site-packages/platformio/commands/platform.pyc'> ['PlatformFactory', 'PlatformManager', '__builtins__', '__doc__', '__file__', '__name__', '__package__', '_get_installed_platform_data', '_get_platform_data', '_get_registry_platform_data', '_get_registry_platforms', '_original_version', '_print_platforms', 'app', 'cli', 'click', 'dirname', 'exception', 'isdir', 'json', 'platform_frameworks', 'platform_install', 'platform_list', 'platform_search', 'platform_show', 'platform_uninstall', 'platform_update', 'print_boards', 'util']
```

ivankravets, 2018-06-18:
Thanks! This is very useful information for me.

ristomatti, 2018-06-18:
Otherwise I could use the RPi 3 for this purpose but it happens to be in an inconvenient place in the apartment serving as a Z-Wave hub. The Odroid sits on my desk where I do all my microcontroller stuff. Besides it's twice as fast so better suited for pretty much everything in general.

ivankravets, 2018-06-18:
What do you see on RPI3 side when type `python -c "from platformio import util; print util.get_systype()"`?

ristomatti, 2018-06-18:
RPi3:
```
$ python -c "from platformio import util; print util.get_systype()"
linux_armv7l
```
C2:
```
$ python -c "from platformio import util; print util.get_systype()"
linux_aarch64
```

ivankravets, 2018-06-18:
> C2:

What is it?

ristomatti, 2018-06-18:
...Odroid C2.

ivankravets, 2018-06-18:
Please replace files in `~/.platformio/packages/tool-pioplus` by files attached in archive. Does it work on C2?
[tool-pioplus.zip](https://github.com/platformio/platformio-core/files/2113115/tool-pioplus.zip)

ristomatti, 2018-06-19:
The situation improved seemingly lot but still crashing though. Or is this perhaps some unrelated issue?
```
odroid@odr01 ~ $ pio remote device list
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/remote.py", line 129, in device_list
    pioplus_call(sys.argv[1:])
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/core.py", line 113, in pioplus_call
    code = subprocess.call([pioplus_path] + args, **kwargs)
  File "/usr/lib/python2.7/subprocess.py", line 523, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 13] Permission denied

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

ivankravets, 2018-06-19:
Do you see any errors when calling directly `~/.platformio/packages/tool-pioplus/pioplus`?

ristomatti, 2018-06-19:
You should be using `tar` for this kind of thing. Of course `pioplus` didn't have the `+x` flag. :)

I'm still getting some errors on missing modules:

```
$ ~/.platformio/packages/tool-pioplus/pioplus
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
ImportError: No module named twisted.internet
```
```
$ pio remote device list
Traceback (most recent call last):
  File "main.py", line 17, in init main
  File "cmd_home.py", line 17, in init cmd_home
ImportError: No module named bs4
```

ivankravets, 2018-06-19:
Great! We are so close! 😊  Need to prepare `pysite` package for `aarch64`. I'll inform here soon.

ristomatti, 2018-06-19:
Great news! This has been some Intensive problem solving, impressed!

ivankravets, 2018-06-20:
Please remove `~/.platformio/packages` directory on aarch64 side and start PIO Remote again. Does it work now?

ristomatti, 2018-06-21:
Seems to work, awesome! Thank you, this issue can be closed.

```
$ pio remote agent start
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
PlatformIO Plus (https://pioplus.com) v1.3.1
2018-06-21 17:01:28 [info] Name: odr01
2018-06-21 17:01:28 [info] Connecting to PIO Remote Cloud
2018-06-21 17:01:28 [info] Successfully connected
2018-06-21 17:01:28 [info] Authenticating
2018-06-21 17:01:28 [info] Successfully authorized
```

Out of curiosity - what was the root cause if I may ask? I thought Python would be a cross platform language but I guess it also has native libraries for performance reasons similarly as Node.js?

ivankravets, 2018-06-21:
> I thought Python would be a cross platform language 

It is. PIO Plus Core is C++/Python extension which we compile to the native binary. In your case, we used standalone aarch64 Linux distribution to provide these binaries.

ivankravets, 2018-06-21:
Ahh.. yes! Thank you so much for the testing! 🍺

closed 2018-06-21 by ivankravets