platformio/platformio-core #1260: lib_extra_dirs conflicting with lib_deps_external

flurl, 2017-12-26:
### Configuration

**Operating system**: Ubuntu 17.04

**PlatformIO Version** (`platformio --version`): 3.5.0rc15

### Description of problem
I have the following platformio.ini which used to work (not sure up to which PlatformIO version).
But now the project doesn't compile anymore. It fails with:

```
In file included from src/main.cpp:2:0:
lib/Input/Input.h:4:24: fatal error: Keypad_I2C.h: No such file or directory
 
*******************************************************************
* Looking for Keypad_I2C.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Keypad_I2C.h"
* Web  > http://platformio.org/lib/search?query=header:Keypad_I2C.h
*
*******************************************************************
 
#include <Keypad_I2C.h>
^
compilation terminated.
```

If I remove 
```
https://github.com/joeyoung/arduino_keypads.git
```
from `lib_deps_external` it works

#### Steps to Reproduce

1. Create platformio.ini
2. Compile

### Actual Results
Keypad_I2C.h not found

### Expected Results
Should compile without errors as before

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
lib_extra_dirs = .piolibdeps/arduino_keypads, lib/Arduino/libraries/

[common]
lib_deps_external =
    ArduinoJson
;    DHTlib
    ESPAsyncTCP
    ESPAsyncWebServer
    Keypad
    NtpClientLib
    PCF8574
    RTC
    Tasker
    Time
    U8g2
;    https://github.com/gmag11/FSBrowserNG.git
    https://github.com/joeyoung/arduino_keypads.git

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
    ${common.lib_deps_external}

[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
lib_deps =
    ${common.lib_deps_external}
```

ivankravets, 2017-12-26:
That was the bug in LDF with the previous PIO Core versions. The current behavior is correct and I agree with this error. You asked PIO Core to build whole `https://github.com/joeyoung/arduino_keypads.git` repo and its libraries even when you don't need them. However, I like your idea with `lib_extra_dirs = .piolibdeps/arduino_keypads`. Good hook :) 

Just need now exclude from build process WHOLE repository using `lib_ignore` option:

```
[env:nodemcuv2]
...
lib_ignore = arduino_keypads
```

Does it work now?

flurl, 2017-12-28:
Thanks a lot. That did it.
For completeness my working platformio.ini:
```
[platformio]
lib_extra_dirs = .piolibdeps/arduino_keypads, lib/Arduino/libraries/

[common]
lib_deps_external =
    ArduinoJson
;    DHTlib
    ESPAsyncTCP
    ESPAsyncWebServer
    Keypad
    NtpClientLib
    PCF8574
    RTC
    Tasker
    Time
    U8g2
;    https://github.com/gmag11/FSBrowserNG.git
    https://github.com/joeyoung/arduino_keypads.git
ignore_libs = arduino_keypads

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps = ${common.lib_deps_external}
lib_ignore = ${common.ignore_libs}

[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
lib_deps = ${common.lib_deps_external}
lib_ignore = ${common.ignore_libs}
```

ivankravets, 2017-12-28:
Thanks for the confirmation!

closed 2017-12-28 by ivankravets