platformio/platformio-core #1262: Allow GDB for native environment tests

twhiston, 2017-12-28:
It would be extremely useful if gdb could be run on native TEST targets so we could more effectively debug during the testing cycle, it seems like most generic code would benefit from this as it would allow you to do a lot of work before even getting near a microprocessor.

ivankravets, 2017-12-28:
Do you mean this https://github.com/platformio/platformio-core/issues/980 ?

twhiston, 2017-12-28:
yeah that is part of it, but i would like to specifically target the unit tests, and not the main(or setup/loop).

ivankravets, 2017-12-29:
You can do that with `native` dev/platform. See example https://github.com/platformio/platformio-examples/tree/develop/unit-testing/calculator

twhiston, 2017-12-29:
I had actually seen that example and tried something similar but could not get any breakpoints to halt execution.

ivankravets, 2017-12-29:
This feature request consists of

* https://github.com/platformio/platformio-core/issues/948
* https://github.com/platformio/platformio-core/issues/980

Am I right? If yes, please close it and subscribe to issues mentioned above.

ivankravets, 2018-02-08:
Duplicate of https://github.com/platformio/platformio-core/issues/948
Duplicate of https://github.com/platformio/platformio-core/issues/980

closed 2018-02-08 by ivankravets