platformio/platformio-core #1263: xtensa esp8266 toolchain on arm64v8 (aarch64) not working

lobradov, 2017-12-28:
What kind of issue is this?

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:
Armbian on Orange Pi Prime (Ubuntu Xenial based): 
```console
$ uname -a 
Linux 88eda1b42c6c 4.14.8-sunxi64 #68 SMP Thu Dec 21 12:52:19 CET 2017 aarch64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial
```
**PlatformIO Version** (`platformio --version`):
```console
$ platformio --version
PlatformIO, version 3.5.0

$ file ~/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-c++
/root/.platformio/packages/toolchain-xtensa/bin/xtensa-lx106-elf-c++: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, for GNU/Linux 2.6.26, BuildID[sha1]=ad8066d871f3e7952a6b38339fb4b9e23a48d4ef, stripped
```

### Description of problem
While PlatformIO itself installs correctly and can initialize a project, it would seems that binaries pulled for arm64v8 (aarch64) are not actually made for this architecture, so they cause exceptions and dump core.

Would be great to have an option to build a toolchain on the spot, rather than pulling a precompiled one.

```console
$ pio run -v
[Thu Dec 28 22:31:32 2017] Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------
Collected 23 compatible libraries
Looking for dependencies...
No dependencies
xtensa-lx106-elf-g++ -o .pioenvs/esp01_1m/src/test.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30500 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -U__STRICT_ANSI__ -I/root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I/root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include -I/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 -I/root/.platformio/packages/framework-arduinoespressif8266/variants/generic -Isrc -Iinclude src/test.cpp
xtensa-lx106-elf-ar rc .pioenvs/esp01_1m/libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pioenvs/esp01_1m/FrameworkArduino/Esp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30500 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -U__STRICT_ANSI__ -I/root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I/root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include -I/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 -I/root/.platformio/packages/framework-arduinoespressif8266/variants/generic /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Esp.cpp
xtensa-lx106-elf-g++ -o .pioenvs/esp01_1m/FrameworkArduino/FS.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -ffunction-sections -fdata-sections -DARDUINO=20300 -DLWIP_OPEN_SRC -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DPLATFORMIO=30500 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -U__STRICT_ANSI__ -I/root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I/root/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip/include -I/root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 -I/root/.platformio/packages/framework-arduinoespressif8266/variants/generic /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/FS.cpp
Illegal instruction (core dumped)
*** [.pioenvs/esp01_1m/libFrameworkArduinoVariant.a] Error 132
Illegal instruction (core dumped)
*** [.pioenvs/esp01_1m/src/test.o] Error 132
Illegal instruction (core dumped)
*** [.pioenvs/esp01_1m/FrameworkArduino/Esp.o] Error 132
Illegal instruction (core dumped)
*** [.pioenvs/esp01_1m/FrameworkArduino/FS.o] Error 132
====================================================== [ERROR] Took 4.31 seconds ======================================================
```

#### Steps to Reproduce
On Orange Pi Prime (or assumably any Armbian/aarch64 board):
1. git clone https://github.com/lobradov/docker-platformio.git
2. cd docker-platformio
3. docker build -t pio-build .
This will pull the latest debian-slim container image, install python, pull PlatformIO and try to compile an empty test.ino sketch. Feel free to ignore docker specifics. 

### Actual Results
On arm64v8 (aarch64), docker build breaks at compiling test.ino:
```console
[...]
Converting test.ino
Illegal instruction (core dumped)
*** Error 132
AssertionError: :
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/root/.platformio/platforms/espressif8266/builder/main.py", line 375:
target_elf = env.BuildProgram()
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 58:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 257:
env.ConvertInoToCpp()
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/piomisc.py", line 54:
return self.process(contents)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
========================== [ERROR] Took 40.62 seconds ==========================
The command '/bin/sh -c python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)" &&     pio settings set enable_telemetry No &&     pio init --board=esp01_1m -d /tmp/bootstrap &&     cd /tmp/bootstrap &&     pio lib --global install PubSubClient &&     pio lib --global install ArduinoJSON &&     pio run &&     rm -rf /var/lib/apt/lists/* &&     rm -rf /tmp/bootstrap' returned a non-zero code: 1
```

### Expected Results
On amd64 (x86_64), build is normal and goes like:
```console
$ docker build -t pio-build .
Sending build context to Docker daemon  75.26kB
Step 1/5 : FROM debian:stable-slim
stable-slim: Pulling from library/debian
[...]
[Thu Dec 28 22:46:30 2017] Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
--------------------------------------------------------------------------------
PlatformManager: Installing espressif8266
Downloading
[...]
Verbose mode can be enabled via `-v, --verbose` option
Converting test.ino
Collected 25 compatible libraries
Looking for dependencies...
No dependencies
Compiling .pioenvs/esp01_1m/src/test.ino.o
Archiving .pioenvs/esp01_1m/libFrameworkArduinoVariant.a
Compiling .pioenvs/esp01_1m/FrameworkArduino/Esp.o
Compiling .pioenvs/esp01_1m/FrameworkArduino/FS.o
Compiling .pioenvs/esp01_1m/FrameworkArduino/HardwareSerial.o
[...]
Linking .pioenvs/esp01_1m/firmware.elf
Building .pioenvs/esp01_1m/firmware.bin
Calculating size .pioenvs/esp01_1m/firmware.elf
text	   data	    bss	    dec	    hex	filename
221014	    888	  29408	 251310	  3d5ae	.pioenvs/esp01_1m/firmware.elf
========================= [SUCCESS] Took 13.27 seconds =========================
 ---> 813b8b1cdc99
Removing intermediate container d99aef256aae
Successfully built 813b8b1cdc99
Successfully tagged pio-build:latest
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
```

**Source file to reproduce issue:**
```cpp
void setup() { }; 
void loop() { };
```

### Additional info

ivankravets, 2017-12-29:
Thanks a lot for the great report!  I've just fixed manifest.

In other words, we will not have espressif8266 toolchain for aarch64. Need to contact Espressif team and ask them about toolchain for aarch64

closed 2017-12-29 by ivankravets

OttoWinter, 2019-01-23:
@ivankravets Bummer, but ok, that's out of our control :(

Though there is a quick fix for this without having a special aarch64 build of the toolchain. Most OSs that at the moment support arm64v8 also have 32-bit compatibility modes for armhf.

I manually tested running the armhf toolchain on a arm64v8 system (HassOS on RPi 3 in this case) and it worked fine 🎉 Would it be possible to tag the linux_armhf builds also as linux_aarch64?

ivankravets, 2019-01-24:
@OttoWinter could you provide an output from your machine:
```
python -c "from platformio import util; print(util.get_systype())"
```

OttoWinter, 2019-01-24:
@ivankravets systype is `linux_aarch64`. However, the toolchain available under `linux_armhf` also works for `linux_aarch64` (at least on an RPi 3, can't comment on other systems)

In my software, I actually ended up just patching the `platformio.util.get_systype` function for my users to return `armhf` for `aarch64` systems, which seems to work fine.

ivankravets, 2019-01-24:
Actually, I don't understand why it works for you and does not for @lobradov.

This is a current manifest:
```json
{
    "description": "xtensa-gcc",
    "name": "toolchain-xtensa",
    "system": [
        "linux_armv6l",
        "linux_armv7l",
        "linux_armv8l"
    ],
    "url": "https://github.com/jcmvbkbc/gcc-xtensa",
    "version": "1.40802.1"
}

```

So, we can add `linux_aarch64`. Does this https://bintray.com/platformio/dl-packages/download_file?file_path=toolchain-xtensa-linux_armv6l-1.40802.1.tar.gz work for you? Please add `linux_aarch64` to `package.json` and move to `~/.platformio/packages`

OttoWinter, 2019-02-14:
Hi @ivankravets ,

I spent some time looking more into this issue. I'm now no longer quite sure why my previous patch worked (as you said correctly, armhf is not in the manifest). However I now have a patch that is really working:

https://github.com/esphome/esphome-docker-base/blob/master/platformio-fix-aarch64.patch

With that aarch64 (or armv8l) builds work correctly and the toolchain runs too. I'm not quite sure what the difference is between armv8l and aarch64, but the armv8l toolchain appears to work for all aarch64 devices I have.

`toolchain-xtensa32` does have `linux_aarch64` in addition to `linux_armv8l` specified ([ref](http://dl.platformio.org/packages/manifest.json)). Would it be possible to add `toolchain_aarch64` to the `toolchain-xtensa-linux_armv6l-1.40802.1.tar.gz` file? Then I wouldn't need to manually apply the patch.

ivankravets, 2019-02-15:
Done! Please run
```
pio update
```

pfeerick, 2019-02-22:
Not sure if this is the right issue to lodge this against (as I thought there was another ESP8266 arm/aarch issue but can't find it)... but on a pine64 Pinebook (aarch) with headmelted's vscode-oss, the toolchain appears to be correct (i.e. the code is all compiling, but falls over when trying to run esptool right at the end. I *think* it is a x86/x64 binary is downloaded instead of a armhf one. I deliberately deleted the tool-esptool folder so platformio had to re-download it when it ran a build...

```
pfeerick@pinebook:~/Documents/PlatformIO/Projects/test$ pio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.181225    [Up-to-date]
Updating contrib-piohome                 @ 2.0.0          [Up-to-date]
Updating tool-pioplus                    @ 2.0.2          [Up-to-date]

Platform Manager
================
Platform Espressif 8266
--------
Updating espressif8266                   @ 2.0.0          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20500.0      [Up-to-date]
Updating toolchain-xtensa                @ 2.40802.190218 [Up-to-date]

Library Manager
===============
pfeerick@pinebook:~/Documents/PlatformIO/Projects/test$ rm -R "~/.platformio/packages/tool-esptool"
pfeerick@pinebook:~/Documents/PlatformIO/Projects/test$

```

```
pfeerick@pinebook:~/Documents/PlatformIO/Projects/test$ pio run -v
Processing oak (platform: espressif8266; board: oak; framework: arduino)
------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-esptool @ <2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/oak.html
PLATFORM: Espressif 8266 > DigiStump Oak
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
No dependencies
<lambda>(["checkprogsize"], [".pioenvs/oak/firmware.elf"])
"esptool" -eo "/home/pfeerick/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" -bo .pioenvs/oak/firmware.bin -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo .pioenvs/oak/firmware.elf -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
MethodWrapper(["checkprogsize"], [".pioenvs/oak/firmware.elf"])
sh: 1: esptool: not found
*** [.pioenvs/oak/firmware.bin] Error 127
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  32.6% (used 26708 bytes from 81920 bytes)
PROGRAM: [===       ]  25.8% (used 269648 bytes from 1044464 bytes)
.pioenvs/oak/firmware.elf  :
section                                                                                     size         addr
.data                                                                                       1272   1073643520
.noinit                                                                                        4   1073644792
.irom0.text                                                                               241724   1075843088
.text                                                                                      26088   1074790400
.rodata                                                                                      564   1073644800
.bss                                                                                       24872   1073645368
.comment                                                                                    5157            0
.xtensa.info                                                                                  56            0
.xt.prop._ZTISt9exception                                                                     12            0
.xt.prop._ZNSt14_Function_baseD2Ev                                                            48            0
.xt.prop._ZNSt8functionIFvvEE4swapERS1_                                                       36            0
.xt.lit._ZN18ScheduledFunctions16ScheduledElementD2Ev                                          0            0
.xt.lit._ZNSt10_List_baseIN18ScheduledFunctions16ScheduledElementESaIS1_EE8_M_clearEv          0            0
.xt.prop._ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE0EED2Ev                             48            0
.xt.prop._ZN18ScheduledFunctions16ScheduledElementD2Ev                                        48            0
.xt.prop._ZNSt10_List_baseIN18ScheduledFunctions16ScheduledElementESaIS1_EE8_M_clearEv        72            0
.xt.prop._ZTIN10__cxxabiv117__class_type_infoE                                                12            0
.xt.prop._ZTVN10__cxxabiv117__class_type_infoE                                                12            0
.xt.prop._ZTIN10__cxxabiv115__forced_unwindE                                                  12            0
.xt.prop._ZTIN10__cxxabiv119__foreign_exceptionE                                              12            0
.xt.prop._ZTIN10__cxxabiv120__si_class_type_infoE                                             12            0
.xt.prop._ZTVN10__cxxabiv120__si_class_type_infoE                                             12            0
.xt.prop._ZTISt9type_info                                                                     12            0
[... cut ...]
Total                                                                                    1046496
======================================== [ERROR] Took 9.33 seconds ========================================
```

```
pfeerick@pinebook:~/.platformio/packages/tool-esptool$ ls -la
total 36
drwx------  2 pfeerick pfeerick  4096 Feb 22 06:41 .
drwxrwxr-x 10 pfeerick pfeerick  4096 Feb 22 06:41 ..
-rwxr-xr-x  1 pfeerick pfeerick 24328 Feb 17  2018 esptool
-rw-r--r--  1 pfeerick pfeerick   258 Feb 20  2018 package.json
```

ivankravets, 2019-02-22:
> tool-esptool

You mean this https://github.com/igrr/esptool-ck/releases

Could someone compile v0.4.13 on aarch64 and share binary? I'll deploy to our CDN. Thanks!

pfeerick, 2019-02-22:
That's the one! I thought I'd saw a post where you'd reqested the binary, but couldn't find it again to follow up on it. :(

I've compiled and tried in on my pinebook, seems to be working great. :)

Binary is downloadable here : https://dl.bintray.com/pfeerick/aarch64/esptool

Compile log: 
```
pfeerick@pinebook:~/repos/esptool-ck$ make
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o argparse/argparse.o argparse/argparse.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o argparse/argparse_binimagecmd.o argparse/argparse_binimagecmd.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o argparse/argparse_commcmd.o argparse/argparse_commcmd.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o argparse/argparse_elfcmd.o argparse/argparse_elfcmd.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o binimage/esptool_binimage.o binimage/esptool_binimage.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o elf/esptool_elf.o elf/esptool_elf.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o elf/esptool_elf_object.o elf/esptool_elf_object.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o espcomm/delay.o espcomm/delay.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o espcomm/espcomm.o espcomm/espcomm.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o espcomm/espcomm_boards.o espcomm/espcomm_boards.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o infohelper/infohelper.o infohelper/infohelper.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o serialport/serialport.o serialport/serialport.c
cc -std=gnu99 -Os -Wall   -Iinfohelper -Ielf -Ibinimage -Iargparse -Iserialport -Iespcomm  -DVERSION=\"0.4.13\"   -c -o main.o main.c
cc argparse/argparse.o argparse/argparse_binimagecmd.o argparse/argparse_commcmd.o argparse/argparse_elfcmd.o binimage/esptool_binimage.o elf/esptool_elf.o elf/esptool_elf_object.o espcomm/delay.o espcomm/espcomm.o espcomm/espcomm_boards.o infohelper/infohelper.o serialport/serialport.o main.o -o esptool
strip esptool 2>/dev/null \
|| -strip esptool
```
SHA256
```
0ec12219d65306025626dd25f13a0881a876e4c8356c14c1065db09953859a25  esptool
```

PIO verbose compile and upload of simple blink test for NODEMCU board
```
Processing oak (platform: espressif8266; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
No dependencies
xtensa-lx106-elf-g++ -o .pioenvs/oak/src/main.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -Wall -DPLATFORMIO=30604 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_NODEMCUV2\" -DLWIP_OPEN_SRC -DTCP_MSS=536 -DVTABLES_IN_FLASH -Isrc -Iinclude -I/home/pfeerick/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I/home/pfeerick/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include -I/home/pfeerick/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 -I/home/pfeerick/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip2/include -I/home/pfeerick/.platformio/packages/framework-arduinoespressif8266/variants/nodemcu src/main.cpp
xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH /home/pfeerick/.platformio/packages/framework-arduinoespressif8266/tools/sdk/ld/eagle.app.v6.common.ld.h -o .pioenvs/oak/ld/eagle.app.v6.common.ld
xtensa-lx106-elf-ar rc .pioenvs/oak/libFrameworkArduinoVariant.a
xtensa-lx106-elf-ranlib .pioenvs/oak/libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pioenvs/oak/FrameworkArduino/Esp-version.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -Wall -DPLATFORMIO=30604 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_NODEMCU -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_NODEMCUV2\" -DLWIP_OPEN_SRC -DTCP_MSS=536 -DVTABLES_IN_FLASH -I/home/pfeerick/.platformio/packages/framework-arduinoespressif8266/tools/sdk/include -I/home/pfeerick/.platformio/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include -I/home/pfeerick/.pl [... cut ...]
................................................................................ [ 64% ]
................................................................................ [ 97% ]
.......                                                                          [ 100% ]
starting app without reboot
espcomm_send_command: sending command header
espcomm_send_command: sending command payload
espcomm_send_command: receiving 2 bytes of data
closing bootloader
============================================== [SUCCESS] Took 62.08 seconds ==============================================
```

ivankravets, 2019-02-23:
@pfeerick Done! Thank you very much!!! 👍