platformio/platformio-core #1268: lib\Timezone-master\Timezone.cpp:45:17: error: 'year' was not declared in this scope

ksperr, 2017-12-31:
### Configuration

I'm using last updated version of platformio and atom.

Compiling for:
Processing nodemcuv2 (platform: espressif8266; upload_port: COM[7]; board: nodemcuv2; framework: arduino)

### Description of problem#### Steps to Reproduce

The problem is when i try to compile a code that uses this library:
https://github.com/JChristensen/Timezone
I placed this libary on the external libraries folder of the project.

> lib\Timezone-master\Timezone.cpp: In member function 'time_t Timezone::toLocal(time_t)':
> lib\Timezone-master\Timezone.cpp:45:17: error: 'year' was not declared in this scope
> if (year(utc) != year(_dstUTC)) calcTimeChanges(year(utc));
> ^

I think its the same that ocurrs when i try to compile with platformio a file that have the loop and the setup before the function declaration, i'm right?
I compiled this project with arduino IDE, and when i tried it with platform io it gives me these declaration errors. I moved loop and setup at the end of the file and some of them are fixed...but is there any possiblity to compile without to change the code of the libraries?

ivankravets, 2018-01-03:
Use `INO` file extension instead of `CPP` or http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

closed 2018-01-03 by ivankravets

ksperr, 2018-01-08:
the main.cpp to main.ino? 
Not fix the problem.

lib\Timezone-master\Timezone.cpp: In member function 'time_t Timezone::toLocal(time_t)':
lib\Timezone-master\Timezone.cpp:45:17: error: 'year' was not declared in this scope
if (year(utc) != year(_dstUTC)) calcTimeChanges(year(utc));
^

LowArt, 2018-01-08:
I have encountered a similar (the same?) issue with the ArduinoIDE.

The error came up when I was working on an old project on my new PC. On my old laptop the project will be compiled without any errors. 

The problem is the current version (2.4.0) of the ESP8266 board drivers. Switching back to the 2.3.0 version solved the problem for me. 

Even so I worked in the past with PlatformI0 I am not certain how to change the setting there. However, great program. Keep the good work up!

[image]

ivankravets, 2018-01-08:
That is easy with PlatformIO:

```ini
[env:...]
platform = espressif8266@1.5.0
...
```

You can use different dev/platforms per different project. No need to uninstall other dev/platforms as Arduino IDE requires.

ivankravets, 2018-01-08:
* https://github.com/platformio/platform-espressif8266/releases
* http://docs.platformio.org/en/latest/projectconf/section_env_general.html#platform

ksperr, 2018-01-10:
ivankravets if you want i can send you the source code i m tring to compile

ivankravets, 2018-01-10:
@ksperr have you tried `platform = espressif8266@1.5.0`?

ksperr, 2018-01-10:
It works, thanks. 
Can i ask what is the problem? Why the old version works and the new one doesnt ?

ivankravets, 2018-01-10:
@ksperr maybe, something was changed in API. See https://github.com/esp8266/Arduino/releases/tag/2.4.0

proddy, 2018-01-13:
I had the same problem with timezone with PlatformIO 1.6.0 and rolled back to 1.5.0. I reckon it has something to do with the latest Arduino Core v2.4.0

wabbitguy, 2018-01-28:
Same boat here. Arduino IDE 1.8.5, ESP 2.3.0 and timezone compiles Perfectly.

Use ESP2.4.0, a whack of timezone library errors.

Something changed.

proddy, 2018-01-28:
@wabbitguy it works for me now. I think the problem was related to https://github.com/JChristensen/Timezone/issues/7 . Using the latest 1.1.1 release of TimeZone fixed my compile errors with ESP 2.4

wabbitguy, 2018-01-30:
@proddy Thanks for the update, I modified the TimeZone.h myself with the suggested TimeLib change after wrestling with for a couple of days. You said you are using the latest 1.1.1 version of TimeZone, but the GitHub only sys 1.1.0 (that has the TimeLib mod)...I'se confused...:-)

ivankravets, 2018-01-30:
@wabbitguy https://github.com/JChristensen/Timezone/releases

proddy, 2018-01-30:
https://github.com/JChristensen/Timezone/releases/tag/1.1.1