platformio/platformio-core #1269: Broken support for uploading prebuilt filesystem to ESP32

huming2207, 2018-01-01:
### Configuration

**Operating system**: macOS 10.13.2

**PlatformIO Version** (`platformio --version`): 3.5.1a1

### Description of problem

When using the latest platformio, the `pio run --target buildfs` or `pio run --target uploadfs` command are broken. The prebuilt filesystem images are not generated. Probably you guys haven't ready to implement yet or somehow it just doesn't work anyway.

#### Steps to Reproduce

Run `pio run --target buildfs` or `pio run --target uploadfs` with any ESP32 target setting.

### Actual Results

Output message:

```
hu@hu-mac  ~/Projects/Enlight > pio run --target uploadfs
[Mon Jan  1 21:36:48 2018] Processing node32s (data_dir: ./uploadfs; framework: arduino; lib_deps: ArduinoJson, Wire, SPI, FS, https://github.com/samguyer/FastLED, https://github.com/me-no-dev/ESPAsyncWebServer; build_flags: !echo "-DENLIGHT_GIT_REV="\"""$(git rev-parse --short HEAD)-$(date '+%Y%m%d-%H%M%S-%Z')@$(hostname)"\""; platform: espressif32; board: node32s)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Detected non-PlatformIO `data_dir` option in `[env:node32s]` section
Verbose mode can be enabled via `-v, --verbose` option
Collected 21 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <ArduinoJson> v5.12.0
|-- <Wire> v1.0
|-- <SPI> v1.0
|-- <FS> v1.0
|-- <FastLED> v3.1.4
|-- <ESP Async WebServer> v1.1.1
|   |-- <AsyncTCP> v1.0.1
|   |-- <FS> v1.0
|   |-- <WiFi> v1.0
|   |-- <ArduinoJson> v5.12.0
|-- <Preferences> v1.0
|-- <SPIFFS> v1.0
|   |-- <FS> v1.0
|-- <ESPmDNS> v1.0
*** Do not know how to make File target `uploadfs' (/Users/hu/Projects/Enlight/uploadfs).  Stop.
================================================================================= [ERROR] Took 1.50 seconds =================================================================================
 ✘ hu@hu-mac  ~/Projects/Enlight> pio run --target buildfs 
[Mon Jan  1 21:37:07 2018] Processing node32s (data_dir: ./uploadfs; framework: arduino; lib_deps: ArduinoJson, Wire, SPI, FS, https://github.com/samguyer/FastLED, https://github.com/me-no-dev/ESPAsyncWebServer; build_flags: !echo "-DENLIGHT_GIT_REV="\"""$(git rev-parse --short HEAD)-$(date '+%Y%m%d-%H%M%S-%Z')@$(hostname)"\""; platform: espressif32; board: node32s)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Detected non-PlatformIO `data_dir` option in `[env:node32s]` section
Verbose mode can be enabled via `-v, --verbose` option
Collected 21 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <ArduinoJson> v5.12.0
|-- <Wire> v1.0
|-- <SPI> v1.0
|-- <FS> v1.0
|-- <FastLED> v3.1.4
|-- <ESP Async WebServer> v1.1.1
|   |-- <AsyncTCP> v1.0.1
|   |-- <FS> v1.0
|   |-- <WiFi> v1.0
|   |-- <ArduinoJson> v5.12.0
|-- <SPIFFS> v1.0
|   |-- <FS> v1.0
|-- <Preferences> v1.0
|-- <ESPmDNS> v1.0
*** Do not know how to make File target `buildfs' (/Users/hu/Projects/Enlight/buildfs).  Stop.
================================================================================= [ERROR] Took 1.29 seconds =================================================================================
 ✘ hu@hu-mac  ~/Projects/Enlight
```

### Expected Results

I don't know cuz I haven't even seen it worked for me.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:node32s]
build_flags = !echo "-DENLIGHT_GIT_REV="\"""$(git rev-parse --short HEAD)-$(date '+%Y%m%d-%H%M%S-%Z')@$(hostname)"\""
platform = espressif32
board = node32s
framework = arduino
data_dir = ./uploadfs
lib_deps =
    ArduinoJson
    Wire
    SPI
    FS
    https://github.com/samguyer/FastLED
    https://github.com/me-no-dev/ESPAsyncWebServer
```

ivankravets, 2018-01-03:
We currently don't support build/upload to FS for ESP32. Please file a feature request here https://github.com/platformio/platform-espressif32/issues

closed 2018-01-03 by ivankravets

huming2207, 2018-01-05:
Hi @ivankravets 

I've submitted this feature request to the platform repo. Please implement it soon if possible. Thanks.

Here's the issue thread: https://github.com/platformio/platform-espressif32/issues/62

Regards,
Jackson