platformio/platformio-core #1272: Home: Could not load recent projects

HollisTech, 2018-01-02:
Server error

There doesn't appear to be any other information.

ivankravets, 2018-01-02:
Please share here `~/.platformio/homestate.json` file. Thanks!

HollisTech, 2018-01-03:
``{"storage": {"homeDir": "C:\\Users\\Mark", "coreCaller": "vscode", "cid": "8dade227-34c2-4810-eb1d-ae85f85bac54", "projectsDir": "C:\\Users\\Mark\\Documents\\PlatformIO\\Projects", "lastCheckPlatformUpdates": 1514933837960, "lastUpdateCorePackages": 1514933837959, "coreVersion": "3.5.0", "routeBadges": {}, "lastCheckLibraryUpdates": 1514933837960, "recentProjects": ["C:\\Users\\Mark\\Documents\\PlatformIO\\Projects\\171209-143700-featheresp32-pwm-test", "C:\\Users\\Mark\\Documents\\PlatformIO\\Projects\\171218-144021-featheresp32-oled", "C:\\Users\\Mark\\Documents\\PlatformIO\\Projects\\esp32-threaded-test"], "coreSystype": "windows_x86", "showOnStartup": {"vscode": false}}, "inputValues": {"libBuiltinFilter": "timer", "boardsFilter": "Adafruit ESP32 Feather"}}

ivankravets, 2018-01-03:
Could you reproduce this issue after IDE restart?

HollisTech, 2018-01-03:
yes.

Mark Roddy

ivankravets, 2018-01-03:
Please open PIO IDE Terminal and type `pio home`. Do you see any errors?

HollisTech, 2018-01-03:
Yes it is the same issue "could not load recent projects" "server error"
[image]

ivankravets, 2018-01-06:
Please share the contents of these files:
1. C:\Users\Mark\Documents\PlatformIO\Projects\171209-143700-featheresp32-pwm-test\platformio.ini
2. C:\Users\Mark\Documents\PlatformIO\Projects\171218-144021-featheresp32-oled\platformio.ini
3. C:\Users\Mark\Documents\PlatformIO\Projects\esp32-threaded-test\platformio.ini

Thanks!

HollisTech, 2018-01-06:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
monitor_baud=115200`
```

They all look pretty much the same as the above, with the same result when opening pio home.

Here's the one I'm currently working on (esp32-threaded) it is slightly different:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
monitor_baud=115200
build_flags=-Wl,-Map,output.map
upload_port=esp32-1.local

;[env:native]
; platform = native
```

ivankravets, 2018-01-06:
How did you fix it?

HollisTech, 2018-01-07:
I accidentally downloaded a repo into .platformio/platforms that should not have been there. Once I moved it out of there all was good. Sorry. My blunder.

closed 2018-01-24 by ivankravets