platformio/platformio-core #1276: Adafruit ATWIN C1500  fails to connect on PIO upload only

TheBeachMaster, 2018-01-05:
### Configuration

**Windows 10 - Microsoft Windows [Version 10.0.16299.125]**:

**PlatformIO Version** (`PlatformIO, version 3.5.0`):

### Description of problem
Running `platformio run --target upload --upload-port COM[NUMBER]` on arduino project attached to a networking module such as Adafruit WiFi ATWINC1500(- observed here first) causes unexpected behavior whereby the module would hang attempting to connect to a network.

However, using other tools such as Visual Studio Code Arduino extension or Arduino IDE or Visual Micro for Arduino on Visual Studio IDE the module works as expected.

This led me to believe that this is a PIO-core issue.

Unfortunately I'm not able to setup a proper debug flag on the WiFi 101 library to enable me to see this more clearly.

Tested on **Arduino Mega 2560**

#### Steps to Reproduce

1. Setup an [Arduino Environment](https://www.arduino.cc/en/Main/Software) 
2. Get the following Libraries [AfricasTalkingIoT](https://github.com/TheBeachMaster/AfricasTalkingIOT/releases/download/v.0.3.61-alpha/AfricasTalkingCloud.zip) and [Wifi101](https://github.com/arduino-libraries/WiFi101/archive/0.15.0.zip) and install
3. Clone the project from [this link](https://github.com/TheBeachMaster/pioremotesample) or copy the following command `git clone https://github.com/TheBeachMaster/pioremotesample.git`. All libraries required to run the application are found in the lib directory no need to add anything new.
4.Using Arduino IDE, with the installed libraries, run the project as is (- Arduino IDE may require the project folder to be changed). **Do not modify the code!!**.
5.Similarly, with Visual Studio Code setup with the Arduino Extension run the project as is.
6. Now run `platformio run --target upload --upload-port COM[NUMBER]` against the project. Replacing `NUMBER` with your **Arduno MEGA** serial port number

### Actual Results
Module Hangs on attempting to connect.

### Expected Results
Module connects successfully to an access point

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino 

upload_port = COM[15]
```

**Source file to reproduce issue:**
```cpp

#include <AfricasTalkingCloud.h>
#include <SPI.h>
#include <WiFi101.h>

char ssid[] = "AfricasTalking IoT GW";     
char pass[] = "africastalking"; 
int status = WL_IDLE_STATUS;
char msg[50];
long value = 0;
long lastMsg = 0;
WiFiClient _client101;
boolean sandbox = true;
void callback(char* topic, byte* payload, unsigned int length);
AfricasTalkingCloudClient client(sandbox, callback, _client101);

void setup() {
	WiFi.setPins(9, 3, 4);
    while (!Serial);
	Serial.begin(9600);
	connectToWPA();
}

// the loop function runs over and over again until power down or reset
void loop() {
	if (!client.connected())
	{
		_keepAlive();
	}
	if (!client.loop()) {
		Serial.print("Client disconnected...");
		if (client.connect("ATWIC1500", "PlatformIO", "pass")) {
			Serial.println("reconnected.");
		}
		else {
			Serial.println("failed.");
		}
	}
	long now = millis();
	if (now - lastMsg > 5000) {
		lastMsg = now;
		++value;
		snprintf(msg, 75, "Data #%ld", value);
		Serial.print("Publish message: ");
		Serial.println(msg);
		client.publish("PlatformIO/ATWIC1500/readings", msg);
	}
  
}

void callback(char * topic, byte * payload, unsigned int length)
{
	Serial.print("Message arrived [");
	Serial.print(topic);
	Serial.print("] ");
	for (unsigned int i = 0; i < length; i++) {
		Serial.print((char)payload[i]);
	}
	Serial.println();
}
void connectToWPA()
{
	Serial.println(F("Setting up connection...."));
	// check for the presence of the shield:
	if (WiFi.status() == WL_NO_SHIELD) {
		Serial.println("WiFi shield not present");
		// don't continue:
		while (true);
	}
	// attempt to connect to Wifi network:
	while (status != WL_CONNECTED) {
		Serial.print("Attempting to connect to WPA SSID: ");
		Serial.println(ssid);
		// Connect to WPA/WPA2 network:
		status = WiFi.begin(ssid, pass);

		// wait 10 seconds for connection:
		delay(10000);
	}
	// you're connected now, so print out the data:
	Serial.print("You're connected to the network");
	printCurrentNet();
	printWifiData();
}

void printWifiData() {
	// print your WiFi shield's IP address:
	IPAddress ip = WiFi.localIP();
	Serial.print("IP Address: ");
	Serial.println(ip);
	Serial.println(ip);

	// print your MAC address:
	byte mac[6];
	WiFi.macAddress(mac);
	Serial.print("MAC address: ");
	Serial.print(mac[5], HEX);
	Serial.print(":");
	Serial.print(mac[4], HEX);
	Serial.print(":");
	Serial.print(mac[3], HEX);
	Serial.print(":");
	Serial.print(mac[2], HEX);
	Serial.print(":");
	Serial.print(mac[1], HEX);
	Serial.print(":");
	Serial.println(mac[0], HEX);

}

void printCurrentNet() {
	// print the SSID of the network you're attached to:
	Serial.print("SSID: ");
	Serial.println(WiFi.SSID());

	// print the MAC address of the router you're attached to:
	byte bssid[6];
	WiFi.BSSID(bssid);
	Serial.print("BSSID: ");
	Serial.print(bssid[5], HEX);
	Serial.print(":");
	Serial.print(bssid[4], HEX);
	Serial.print(":");
	Serial.print(bssid[3], HEX);
	Serial.print(":");
	Serial.print(bssid[2], HEX);
	Serial.print(":");
	Serial.print(bssid[1], HEX);
	Serial.print(":");
	Serial.println(bssid[0], HEX);

	// print the received signal strength:
	long rssi = WiFi.RSSI();
	Serial.print("signal strength (RSSI):");
	Serial.println(rssi);

	// print the encryption type:
	byte encryption = WiFi.encryptionType();
	Serial.print("Encryption Type:");
	Serial.println(encryption, HEX);
	Serial.println();
}
//Reconnect always
void _keepAlive() {

	while (!client.connected()) {
		// Connect to WPA...
		connectToWPA();
		Serial.print(F("Contacting Server: "));
		Serial.println();
		if (client.connect("ATWIC1500", "PlatformIO", "Password")) {
			Serial.println("connected");
			client.publish("PlatformIO/ATWIC1500/birth", "ATWIC1500");
			client.subscribe("PlatformIO/ATWIC1500/commands");
		}
		else {
			Serial.print("Failed, rc=");
			Serial.print(client.state());
			Serial.println(" Retrying...");
			delay(6000);
		}
	}
}
```

### Additional info 
At the time of writing the Africa's Talking IoT library had not been published on Platform IO library registry yet, get it from the provided link.

No confidential information has been sent on the files, the MQTT broker allows anonymous connections at the time of this writing, if connections fail due to authentication  : username should be set to `sandbox` password `sandbox` topics must begin with `sandbox/`

For the module : the  CS pin is connected to Arduino PIN9 , IRQ is connected to Arduino PIN3 , and RST pin is connected to PIN4.  
More details on the module can be obtained from [this link](https://learn.adafruit.com/adafruit-atwinc1500-wifi-module-breakout/wiring-and-test)

TheBeachMaster, 2018-01-09:
This is very interesting. 
I added the `false` flag to `lib_archive` on a `c++` file and it worked really well however the problem still occurs on a `.ino` file. 

## Contents of .cpp file 
```c++ 
// #include <Arduino.h>
#include <AfricasTalkingCloud.h>
#include <SPI.h>
#include <WiFi101.h>

char ssid[] = "Techies@WorkAgain";
char pass[] = "ccad32da7b5d8";
int  status = WL_IDLE_STATUS;
char message[50];
long value = 0, lastMessage = 0;

WiFiClient atwinc1500;
boolean sandbox = true;
void sandboxCallback(char topic [], byte* payload, unsigned int length);

AfricasTalkingCloudClient client(sandbox, sandboxCallback, atwinc1500);
void _keepAlive();
void connectToWPA();
void printWifiData();
void printCurrentNet();

void setup()
{
    WiFi.setPins(9, 3, 4);
    while(!Serial);
    Serial.begin(9600);
    connectToWPA();
}

void loop()
{
	if (!client.connected())
	{
		_keepAlive();
	}
	if (!client.loop()) {
		Serial.print("Client disconnected...");
		if (client.connect("ATWIC1500", "PlatformIO", "pass")) {
			Serial.println("reconnected.");
		}
		else {
			Serial.println("failed.");
		}
	}
	long now = millis();
	if (now - lastMessage > 5000) {
		lastMessage = now;
		++value;
		snprintf(message, 75, "Data #%ld", value);
		Serial.print("Publish message: ");
		Serial.println(message);
		client.publish("PlatformIO/ATWIC1500/readings", message);
	}
}

void sandboxCallback(char topic [], byte* payload, unsigned int length)
{
    Serial.println("Message arrived [");
    Serial.print(topic);
    Serial.print("]");
    for (unsigned int i = 0; i < length; i++)
    {
        Serial.println((char)payload[i]);
    }
    Serial.println();
}
//Reconnect always
void _keepAlive() {

	while (!client.connected()) {
		// Connect to WPA...
		connectToWPA();
		Serial.print(F("Contacting Server: "));
		Serial.println();
		if (client.connect("ATWIC1500", "PlatformIO", "Password")) {
			Serial.println("connected");
			client.publish("PlatformIO/ATWIC1500/birth", "ATWIC1500");
			client.subscribe("PlatformIO/ATWIC1500/commands");
		}
		else {
			Serial.print("Failed, rc=");
			Serial.print(client.state());
			Serial.println(" Retrying...");
			delay(6000);
		}
	}
}

void connectToWPA()
{
	Serial.println(F("Setting up connection...."));
	// check for the presence of the shield:
	if (WiFi.status() == WL_NO_SHIELD) {
		Serial.println("WiFi shield not present");
		// don't continue:
		while (true);
	}
	// attempt to connect to Wifi network:
	while (status != WL_CONNECTED) {
		Serial.print("Attempting to connect to WPA SSID: ");
		Serial.println(ssid);
		// Connect to WPA/WPA2 network:
		status = WiFi.begin(ssid, pass);

		// wait 10 seconds for connection:
		delay(10000);
	}
	// you're connected now, so print out the data:
	Serial.print("You're connected to the network");
	printCurrentNet();
	printWifiData();
}

void printWifiData() {
	// print your WiFi shield's IP address:
	IPAddress ip = WiFi.localIP();
	Serial.print("IP Address: ");
	Serial.println(ip);
	Serial.println(ip);

	// print your MAC address:
	byte mac[6];
	WiFi.macAddress(mac);
	Serial.print("MAC address: ");
	Serial.print(mac[5], HEX);
	Serial.print(":");
	Serial.print(mac[4], HEX);
	Serial.print(":");
	Serial.print(mac[3], HEX);
	Serial.print(":");
	Serial.print(mac[2], HEX);
	Serial.print(":");
	Serial.print(mac[1], HEX);
	Serial.print(":");
	Serial.println(mac[0], HEX);

}

void printCurrentNet() {
	// print the SSID of the network you're attached to:
	Serial.print("SSID: ");
	Serial.println(WiFi.SSID());

	// print the MAC address of the router you're attached to:
	byte bssid[6];
	WiFi.BSSID(bssid);
	Serial.print("BSSID: ");
	Serial.print(bssid[5], HEX);
	Serial.print(":");
	Serial.print(bssid[4], HEX);
	Serial.print(":");
	Serial.print(bssid[3], HEX);
	Serial.print(":");
	Serial.print(bssid[2], HEX);
	Serial.print(":");
	Serial.print(bssid[1], HEX);
	Serial.print(":");
	Serial.println(bssid[0], HEX);

	// print the received signal strength:
	long rssi = WiFi.RSSI();
	Serial.print("signal strength (RSSI):");
	Serial.println(rssi);

	// print the encryption type:
	byte encryption = WiFi.encryptionType();
	Serial.print("Encryption Type:");
	Serial.println(encryption, HEX);
	Serial.println();
}
``` 
The source code is similar to the `.ino` file 

## Contents of `.ini` file 
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino 
lib_archive = false
upload_port = COM[15]
  
Code repository can be found at `https://github.com/TheBeachMaster/pioremotesample`. 

Edit wifi configs by changing these lines 
```c++  
char ssid[] = "Techies@WorkAgain";
char pass[] = "ccad32da7b5d8";
``` 
Then run the command `platformio run --target upload --upload-port COM[NUMBER]` 

## Expected results 
`**Code works fine**` 

## Issues 
`.ino` code fails 

## Additional info 
`.ino` code is found under the `asset` folder can be copied to the `src` folder and tested

ivankravets, 2018-01-15:
Please specify a pointer to callback
```c
AfricasTalkingCloudClient client(sandbox, &callback, _client101);
```

closed 2018-01-15 by ivankravets