platformio/platformio-core #1279: Home: Libraries: Could not install library

MichaelSasser, 2018-01-05:
> PIO Core Call Error: Library Storage: C:\Users\Username\.platformio\lib
> LibraryManager: Installing SparkFun_BME280_Arduino_Library
> git version 2.13.2
> 
> Cloning into 'C:\Users\Username\.platformio\lib\_tmp_installing-gelahr-package'...
> fatal: Invalid path '/cygdrive/c/Users/Username/.platformio/lib/_tmp_installing-gelahr-package/C:\Users\Username\.platformio\lib\_tmp_installing-gelahr-package': No such file or directory
> Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git', 'C:\\Users\\Username\\.platformio\\lib\\_tmp_installing-gelahr-package']

Reproducible after restarting VSC.
The Input was: https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git

ivankravets, 2018-01-06:
Could you open VSC Terminal and type `pio lib -g install https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git`. Does it work?

MichaelSasser, 2018-01-07:
No, I also deactivated my antivirus software for testing.

The lib dir exists:
```
$ ls -lah /cygdrive/c/Users/Username/.platformio/lib/
insgesamt 16K
drwx------+ 1 Username Kein 0  7. Jan 09:01 ./
drwx------+ 1 Username Kein 0  7. Jan 08:06 ../
drwx------+ 1 Username Kein 0  5. Jan 11:48 ArduinoJson_ID64/
drwx------+ 1 Username Kein 0  3. Jan 10:47 PubSubClient_ID89/
```

And git is doing its thing:

```
$ git clone https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git
Klone nach 'SparkFun_BME280_Arduino_Library' ...
remote: Counting objects: 82, done.
remote: Total 82 (delta 0), reused 0 (delta 0), pack-reused 82
Entpacke Objekte: 100% (82/82), Fertig.
```

Output of your command, tested with git 2.13.2 (cygwin):

```
Windows PowerShell
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

PS D:\CurrentDirectory>pio lib -g install https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git
Library Storage: C:\Users\Username\.platformio\lib
LibraryManager: Installing SparkFun_BME280_Arduino_Library
git version 2.13.2
Cloning into 'C:\Users\Username\.platformio\lib\_tmp_installing-bvgwqr-package'...
fatal: Invalid path '/cygdrive/c/Users/Username/.platformio/lib/_tmp_installing-bvgwqr-package/C:\Users\Username\.platformio\lib\_tmp_installing-bvgwqr-package': No such file or directory
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git', >'C:\\Users\\Username\\.platformio\\lib\\_tmp_installing-bvgwqr-package']
PS D:\CurrentDirectory>
```

Now with an updated version of git (2.15.1 cygwin)

```
Windows PowerShell
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

PS D:\CurrentDirectory> pio lib -g install https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git
Library Storage: C:\Users\Username\.platformio\lib
LibraryManager: Installing SparkFun_BME280_Arduino_Library
git version 2.15.1
Cloning into 'C:\Users\Username\.platformio\lib\_tmp_installing-niy1tr-package'...
fatal: Invalid path '/cygdrive/c/Users/Username/.platformio/lib/_tmp_installing-niy1tr-package/C:\Users\Username\.platformio\lib\_tmp_installing-niy1tr-package': No such file or directory
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git', >'C:\\Users\\Username\\.platformio\\lib\\_tmp_installing-niy1tr-package']
PS D:\CurrentDirectory>
```

alexiszecharies, 2018-01-10:
same problem

ivankravets, 2018-01-10:
Could you switch to `cmd.exe`? Maybe, the issue is linked with PIO+Cygwin?

ivankravets, 2018-01-10:
Nevertheless, I recommend using `lib_deps` option in `platformio.ini`:
```
[env:..]
platform = ...
lib_deps = 
  https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git
```

MichaelSasser, 2018-01-13:
Same Problem with `cmd.exe`.  And with `lib_deps`:

`title="[PlatformIO IDE] PlatformIO: IntelliSense Index failed: Cloning into 'd:\PathToProjectDir\.piolibdeps_tmp_installing-tsvpnn-package'...fatal: Invalid path '/cygdrive/d/PathToProjectDir/.piolibdeps/_tmp_installing-tsvpnn-package/d:\PathToProjectDir\.piolibdeps_tmp_installing-tsvpnn-pack"`

And in the VSC Debugging console:

`console.ts:123 [Extension Host] Cloning into 'd:\PathToProjectDir\.piolibdeps\_tmp_installing-tsvpnn-package'...
fatal: Invalid path '/cygdrive/d/PathToProjectDir/.piolibdeps/_tmp_installing-tsvpnn-package/d:\PathToProjectDir\.piolibdeps\_tmp_installing-tsvpnn-package': No such file or directory
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git', 'd:\\PathToProjectDir\\.piolibdeps\\_tmp_installing-tsvpnn-package']`

Installing Libs from PIOs Libraries works like a charm. When I clone `https://github.com/sparkfun/SparkFun_BME280_Arduino_Library.git` by myself and "install" the lib in PIO by the path to the directory it just works. So its not a big deal.

> Maybe, the issue is linked with PIO+Cygwin?

That is possible. I think Cygwin understands only PosixPath and not WindowsPath. Maybe there is somethin mixed up.

ivankravets, 2018-02-03:
Could you reproduce this issue with the latest `$ pio upgrade --dev` version?

ivankravets, 2018-02-13:
Please reopen if you still need help

closed 2018-02-13 by ivankravets

afaehnrich, 2019-09-11:
I had the same problem on a windows machine.
git is adding a windows path (d:\PathToProjectDir\...') to a cygwin path ('/cygdrive/...')
Problem was that I had cygwin installed and  set in my environment PATH variable.
Platformio was using this cygwin git version, but is expecting the native windows version of git.

I removed the path to cygwin from windows PATH and added the path to windows git instead (C:\Program Files\Git\bin).
before this change, when  I entered "git --version" on cmd it showed me "git version 2.16.0"
now it shows me "git version 2.16.0.windows.2".

Everything is working fine.