platformio/platformio-core #1281: pio lib update does not follow semver complaint version constraints.

euphi, 2018-01-06:
As written in https://github.com/marvinroger/homie-esp8266/issues/470, `platformio lib update` updates locally installed libs even if the latest version does not match the semver compliant version match string.

E.g. if there is a dependency to lib@^1.0.0 a lib from the 1.0.x branch is installed initially, but later is updated to 1.1.x, when `pio lib update` is performed.

--> `pio lib update` shall follow the semver dependencies when looking for new libs.

BTW: I thnik this should not apply for global update (`pio lib -g update`) because pio can't know all locally installed projects and a project can install a older revision of a dependency locally.

ivankravets, 2020-09-02:
Please re-test with `pio upgrade --dev`.

closed 2020-09-02 by ivankravets

euphi, 2020-11-10:
Thanks! Seems to run fine now.

commit fe4112a2 (2020-09-02): Follow SemVer complaint version constraints when checking library updates // Resolve #1281