platformio/platformio-core #1283: New ESP8266 v1.6 toolchain has a broken LDF

TheBeachMaster, 2018-01-08:
What kind of issue is this?

- [ ] PlatformIO Core.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu Trusty

**PlatformIO Version** (`PlatformIO 3.5.0`):

### Description of problem
ESP8266 LDF is broken on version 1.6.0. 
The dependency graph is not able to properly resolve all dependencies with this particular update.
In my case the library `SPI.h` is being referenced from the wrong place.
#### Steps to Reproduce

1. Fork the repository `https://github.com/TheBeachMaster/AfricasTalkingIOT`
2. Setup a Travis.org account
3. Add the forked repo and build the `dev` branch

### Actual Results
ESP8266 version: `espressif8266 @ 1.6.0`
`ESP8266` sample will fail as shown below 
[image] 
[image]

Actual build log can be viewed from [this link](https://travis-ci.org/TheBeachMaster/AfricasTalkingIOT/jobs/326319374)

### Expected Results
A successful build on `espressif8266 @ 1.5.0`
[image]

Successful build can be viewed from [this link](https://travis-ci.org/TheBeachMaster/AfricasTalkingIOT/jobs/319853786)

ivankravets, 2018-01-08:
Please read about [Dependency Finder Mode](http://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode) and take a look at default mode.

In your case, 3-rd party library depends on external SPI library. You can resolve that via
```
pio ci --project-option="lib_deps=SPI"
```

Also, please read about http://docs.platformio.org/en/latest/librarymanager/index.html

You have a lot of lines with downloading/unpacking. Do that with PIO:

```
pio lib --global install https://github.com/TheBeachMaster/AfricasTalkingIOT/releases/download/v0.3.6-alpha/AfricasTalkingCloud.zip
....
```

Or better, add these libraries to project dependency list:

```
pio ci --project-option="lib_deps=https://github.com/TheBeachMaster/AfricasTalkingIOT/releases/download/v0.3.6-alpha/AfricasTalkingCloud.zip, Adafruit CC3000 Library, SPI"
```

closed 2018-01-08 by ivankravets

TheBeachMaster, 2018-01-09:
Hey @ivankravets  
I managed to resolve this by adding the `--project-option="lib_ignore=Adafruit-CC3000"` flag against CC3000 and it worked.

I'm not entirely conversant with the `lib_deps` flag, but I'm sure I'll get more adventurous and properly use it. 

Thanks for the help 👍