platformio/platformio-core #1288: Odd red text regarding numerous python 2.7 problems after uploading 'Arduino' based project to ESP32

g01d10x, 2018-01-10:
Hello PlatformIO .. very cool system here.

I'm a bit of a Noob to this environment, but have been tinkering with Arduino and IOS programming for a few years. The other day I installed Platform IO and have started to fall in love with this for programming my ESP32 module .. I like that it's open and hackable and the community behind this seems to be thriving.

I installed Python 3.6 in the hopes of updating my Python 2.7 .. this may have been a mistake. Since then I have been getting these errors, but the odd thing is that they appear AFTER a good compile and install on the ESP32 module. Can anyone help ?

I copied this out of my post-install error messages:

`
19.9 s
platformio run --target upload
Compressed 3072 bytes to 122...
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 1652.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 7462.6 kbit/s)...
Hash of data verified.
Compressed 156384 bytes to 81778...

Wrote 156384 bytes (81778 compressed) at 0x00010000 in 7.2 seconds (effective 172.8 kbit/s)...
Hash of data verified.
 
Leaving...
Hard resetting...
========================= [SUCCESS] Took 18.64 seconds =========================
 
[Wed Jan 10 15:59:17 2018] Processing custom_monitor_port (monitor_port: /dev/cu.SLAB_USBtoUART; platform: espressif32; monitor_b
aud: 115200)
--------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 109, in cli
    result = (envname, ep.process())
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 196, in process
    result = self._run()
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run.py", line 298, in _run
    return p.run(build_vars, build_targets, self.silent, self.verbose)
  File "/Users/bob/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 346, in run
    self.configure_default_packages(variables, targets)
  File "/Users/bob/.platformio/platforms/espressif32/platform.py", line 21, in configure_default_packages
    if "arduino" in variables.get("pioframework"):
TypeError: argument of type 'NoneType' is not iterable
 
============================================================
 
An unexpected error occurred. Further steps:
 
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
 
* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html
 
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
 
============================================================
 
 
Linter
Severity 	Provider 	Description	Line `

ivankravets, 2018-01-10:
Your `platformio.ini` file is broken. Please specify `framework = arduino` and correct INI syntax.

closed 2018-01-10 by ivankravets

g01d10x, 2018-01-11:
Thanks very much .. tried a few new examples and the .ino file seems to be the only difference. Cheers :)