platformio/platformio-core #1297: update lock

mrred128, 2018-01-17:
What kind of issue is this?

- [* ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem
PlatformIO can update itself to an unusable state due to new bugs.

### Additional info

A feature to lock the current state, with the exception of downloading missing modules (platforms, tools) would allow the product to be used in a stable productive environment (enterprise) without going days / weeks of downtime.

ivankravets, 2018-01-17:
You can lock all dependencies in PlatformIO, including, libraries and platforms. See 

* http://docs.platformio.org/en/latest/projectconf/section_env_general.html#platform
* http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps

ivankravets, 2018-01-20:
Please reopen if you still need help

closed 2018-01-20 by ivankravets

trogper, 2024-03-11:
@ivankravets 
could you please update the links? they dont work anymore

ivankravets, 2024-03-12:
- https://docs.platformio.org/en/latest/projectconf/sections/env/options/platform/platform.html
- https://docs.platformio.org/en/latest/projectconf/sections/env/options/library/lib_deps.html