platformio/platformio-core #1299: QtCreator project file contains double hashes ## triggering parsing error

KarlZeilhofer, 2018-01-17:
### Configuration

**Operating system**: Linux Mint 18.1

**PlatformIO Version** (`platformio --version`): 3.5.0

### Description of problem
I've created a project for Qt Creator with `platformio init --ide qtcreator`  
Then the project file contains lines with double hashes, which starts a comment. Therefore the closing double quote isn't detected by the Qt Creator and it complains in Tab 7 (General Messages) about missing closing quotes:
> mypath/platformio.pro(78): Missing closing " quote

Example of problematic line:
`DEFINES += "__INTMAX_C(c)=c ## LL"`

#### Steps to Reproduce

1. create empty project with `platformio init --ide qtcreator --board nodemcuv2`
2. open the .pro file in qtcreator
3. wait for the general messages to appear in tab 7 of qt creator

### Actual Results
[platformio.pro.zip](https://github.com/platformio/platformio-core/files/1640636/platformio.pro.zip)

```
/home/karl/temp/testpro/platformio.pro(62): Missing closing " quote
/home/karl/temp/testpro/platformio.pro(94): Missing closing " quote
/home/karl/temp/testpro/platformio.pro(143): Missing closing " quote
/home/karl/temp/testpro/platformio.pro(147): Missing closing " quote
/home/karl/temp/testpro/platformio.pro(162): Missing closing " quote
/home/karl/temp/testpro/platformio.pro(227): Missing closing " quote
```
Header files are not browsable, because of these errors. 

### Expected Results
No Errors. Browsable header files.

### Workaround
delete the problematic lines from the .pro file and save it.  

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

```

### Additional info
Qt Creator Version: 3.5.1

ivankravets, 2018-01-17:
Please run `pio upgrade --dev` and re-init project with `pio init --ide qtcreator`.

Does it work now?

closed 2018-01-17 by ivankravets

KarlZeilhofer, 2018-01-18:
Yes, thanks!

Unfortunately running only `platformio init` didn't solve the problem, but `pio init --ide qtcreator` as you suggested works fine. Also `platformio init --ide qtcreator --board nodemcuv2` works fine now!

ivankravets, 2018-01-18:
Thanks for the confirmation!

commit bc18941e (2018-01-17): Fix project generator for Qt Creator IDE // Resolve #1299 Resolve #1290