platformio/platformio-core #1300: Unable to build SPIFFS for ESP8266 d1_pro_mini

VonLatvala, 2018-01-18:
What kind of issue is this?

- [x] PlatformIO Core.

Unable to build SPIFFS for ESP8266 d1_pro_mini

------------------------------------------------------------------

### Configuration

**Operating system**: MacOS High Sierra 10.13.2 (17C88)

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.0

### Description of problem

The process of building the SPI FFS fails. The error message is as follows:

```
libc++abi.dylib: terminating with uncaught exception of type std::length_error: vector
*** [.pioenvs/d1_mini_pro/spiffs.bin] Error -6
```

#### Steps to Reproduce

1. `git clone git@bitbucket.org:VonLatvala/esp8266_platformio_spiffs_problem.git`
2. `pio run --target buildfs`

### Actual Results

```
libc++abi.dylib: terminating with uncaught exception of type std::length_error: vector
*** [.pioenvs/d1_mini_pro/spiffs.bin] Error -6
```

No FS image was built.

### Expected Results

FS image built successfully.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
```

**Source file to reproduce issue:**
I have a blank main.cpp with setup&loop in the git repository. Should not matter.

### Additional info

ivankravets, 2018-01-18:
This is a bug, please move it to https://github.com/platformio/platform-espressif8266/issues

closed 2018-01-18 by ivankravets

ivankravets, 2018-01-18:
Fixed in https://github.com/platformio/platform-espressif8266/commit/c84e677a00939ea7d10f8e62450a19f50fca1200