platformio/platformio-core #1304: Home: Could not install platform

ovsel43, 2018-01-21:
PIO Core Call Error: PlatformManager: Installing ststm32
Downloading...
Unpacking...
ststm32 @ 3.8.0 has been successfully installed!
PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.40803.0,<1.40805.0

Downloading...
Unpacking...
Warning! Package Mirror: [Error 5] 
Looking for another mirror...
Downloading...
Unpacking...
Warning! Package Mirror: [Error 5] 
Looking for another mirror...

Error: Could not install 'toolchain-gccarmnoneeabi' with version requirements '>=1.40803.0,<1.40805.0' for your system 'windows_amd64'.
More details: http://bit.ly/faq-package-manager

ivankravets, 2018-01-22:
Have you tried all solutions form http://bit.ly/faq-package-manager?

ovsel43, 2018-01-23:
I tried, to no avail.

The problem was solved after changing CLANG version 64-bit to 86-bit.

ivankravets, 2018-01-24:
Hm... This issue should not be linked with Clang. Nevertheless, that is good that you found a solution.

edogaldo, 2018-01-25:
Hi, I just installed PlatformIO over VSCode over Win10/64bit and I'm having the same error.
I'm not familiar with CLANG so I don't know how to change it.
Could you help me to solve the problem?

Thanks in advance, E.

ivankravets, 2018-01-25:
@edogaldo Have you tried all solutions form http://bit.ly/faq-package-manager?

ovsel43, 2018-01-25:
Hi

http://docs.platformio.org/en/latest/ide/atom.html#ii-clang-for-intelligent-code-completion

1. Deinstall CLANG x64 version (LLVM)

2. Install  CLANG x86 version

At me after that the error has disappeared.

edogaldo, 2018-01-25:
@ivankravets: I tried solution 3 as suggested but no luck, still same error..

@ovsel43: looks like Platformio@VSCode is not using Clang, does it?

ovsel43, 2018-01-26:
Yes Clang not using, can not install it.
But in my case, with the Clang x64 installed, stably produces an error (see the picture): could not install toolchain-gccarmnoneeabi. I tried to install the package STM32 several times.
[image]
I do not know what causes it (Maybe just a coincidence). I do not have time to understand.
Without CLANG or with the installed CLANG x86 the installation is successful (disable the antivirus). I checked several times (once there was a problem with downloading the package).
[image]
Most likely, the problem is not in "CLANG".
Perhaps the problem is in the antivirus (I'm using Kaspersky Endpoint Security) or another.
But during the installation of stm32 the antivirus is turned off.

edogaldo, 2018-01-26:
I give up: I can't disable the antivirus because it is locked at corporate level and there and I can't get STM32 platform working..

@ivankravets: here the error screenshot, hope this helps
[image]

Best regards, E.

edogaldo, 2018-01-26:
Maybe it depends on the fact that there is a strict requirement for a win_64bit version of the "toolchain-gccarmnoneeabi" package and it [does not esists](https://answers.launchpad.net/gcc-arm-embedded/+question/663539)?!

> Tejas Belagod (belagod-tejas) said on 2018-01-25:   #2  
> 
> Hi,
> 
> We don't build a 64-bit Windows version of the toolchain yet. I suspect you would need to start with replacing i686-w64-mingw32-gcc with x86_64-w64-mingw32-gcc and then take it from there.
> 
> Thanks,
>  Tejas.
>

ivankravets, 2018-01-26:
@edogaldo 
> @ivankravets: I tried solution 3 as suggested but no luck, still same error..

Did you use SYSTEM TERMINAL? Not VSCode!

edogaldo, 2018-01-26:
> Did you use SYSTEM TERMINAL? Not VSCode!
yes, I did.

edogaldo, 2018-01-26:
@ivankravets 
Maybe it depends on the fact that there is a strict requirement for a win_64bit version of the "toolchain-gccarmnoneeabi" package and it [does not esists](https://answers.launchpad.net/gcc-arm-embedded/+question/663539)?!

> Tejas Belagod (belagod-tejas) said on 2018-01-25:   #2  
> 
> Hi,
> 
> We don't build a 64-bit Windows version of the toolchain yet. I suspect you would need to start with replacing i686-w64-mingw32-gcc with x86_64-w64-mingw32-gcc and then take it from there.
> 
> Thanks,
>  Tejas.
>

edogaldo, 2018-01-26:
Ok, solution 4 worked!
Could successfully manually install the toolchain using package "toolchain-gccarmnoneeabi-windows-1.40804.0.tar.gz" in https://bintray.com/platformio/dl-packages/toolchain-gccarmnoneeabi/1.40804.0#files
Anyway automatic download is not working in any way..

Best, E.

ivankravets, 2018-02-03:
> Ok, solution 4 worked!

Thanks!

closed 2018-02-03 by ivankravets