platformio/platformio-core #1305: Home: Internal Store Exception

Aukstkalnis, 2018-01-22:
TypeError: Cannot read property 'name' of null
    at http://127.0.0.1:8010/18cd4320383b16562325.min.js:11:33449
    at Array.map (native)
    at _ (http://127.0.0.1:8010/18cd4320383b16562325.min.js:11:33320)
    at http://127.0.0.1:8010/18cd4320383b16562325.min.js:11:33839
    at Array.map (native)
    at y (http://127.0.0.1:8010/18cd4320383b16562325.min.js:11:33779)
    at Function.a [as mapToProps] (http://127.0.0.1:8010/18cd4320383b16562325.min.js:31:330616)
    at r (http://127.0.0.1:8010/18cd4320383b16562325.min.js:6:10642)
    at l (http://127.0.0.1:8010/18cd4320383b16562325.min.js:6:12030)
    at c (http://127.0.0.1:8010/18cd4320383b16562325.min.js:6:12148)

ivankravets, 2018-01-24:
Could you provide more details how to reproduce this issue?

ivankravets, 2018-02-14:
@cpo @Aukstkalnis please open PIO IDE Terminal and type:
```
pio home --port 8888
```

Do you see any errors?

ivankravets, 2018-03-06:
@cpo @Aukstkalnis @MariusLaurinaitis please open PIO IDE Terminal and type:
```
pio home --port 8888
```

Do you see any errors?

Aukstkalnis, 2018-03-06:
After this command I got an error:

Please wait while upgrading PlatformIO...
Error: Traceback (most recent call last):
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\click\core.py", line 1024, in invoke
    Command.invoke(self, ctx)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\platformio\__main__.py", line 74, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\platformio\platformio\maintenance.py", line 44, in on_platformio_start
    after_upgrade(ctx)
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\platformio\platformio\maintenance.py", line 189, in after_upgrade
    app.clean_cache()
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\platformio\platformio\platformio\app.py", line 269, in clean_cache
    with ContentCache() as cc:
  File "c:\users\aukstkalnis.de\.platformio\penv\lib\site-packages\platformio\platformio\platformio\app.py", line 144, in __init__
    self.cache_dir = cache_dir or util.get_cache_dir()
AttributeError: 'module' object has no attribute 'get_cache_dir'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2018-03-06:
http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system ?

Aukstkalnis, 2018-03-06:
After removing platformio core and vscode extension and installing everything again, eferything started to work again. After sending your command no errors occurred.

ivankravets, 2018-03-08:
Thanks for the report!

closed 2018-03-08 by ivankravets