platformio/platformio-core #1308: "pio init" generates project's absolute path in INCLUDEPATH lines for QtCreator

hongquan, 2018-01-24:
What kind of issue is this?

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 17.10

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.2a2

### Description of problem

When generating .pro file for QtCreator, `pio` still leaves absolute path to current project. This is not good if the project is shared among many developers. 

#### Steps to Reproduce

1. Create a folder for project, say `~/Works/AgriConnect/Hardware/Code-Arduino/ebee-generic`.
2. Run `pio init --ide qtcreator --board esp07`
3. In _platformio.ini_, add:

```ini
lib_deps =
  PubSubClient
  OneWire
  SHT1x
  DallasTemperature
```

4. In the project folder, run `pio lib install`.
5. Run `pio init --ide qtcreator` again.

### Actual Results
The _platformio.pro_ file contains project's absolute path

```pro
INCLUDEPATH += "$${HOMEDIR}/Works/AgriConnect/Hardware/Code-Arduino/ebee-generic/src"
INCLUDEPATH += "$${HOMEDIR}/Works/AgriConnect/Hardware/Code-Arduino/ebee-generic/.piolibdeps/PubSubClient_ID89/src"
```

### Expected Results

The _platformio.pro_ file should contain INCLUDEPATH lines like this:

```pro
INCLUDEPATH += "src"
INCLUDEPATH += ".piolibdeps/PubSubClient_ID89/src"
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp07]
platform = espressif8266
board = esp07
framework = arduino
```

ivankravets, 2018-01-24:
>  This is not good if the project is shared among many developers.

You don't need to share this file among developers. PIO will recreate/update it with `pio init` command. The better is to provide comprehensive `platformio.ini` with all dependencies. Later, other develoeprs could use your project with any IDE.

ivankravets, 2018-02-13:
Please add this file to ignore list and PIO will re-create it automatically with `pio init` and fill valid data

closed 2018-02-13 by ivankravets