platformio/platformio-core #1310: Home: Could not load mDNS services

robbor1111, 2018-01-25:
PIO Core Call Error: Extra data: line 3 column 1 - line 3 column 154 (char 415 - 568)

ivankravets, 2018-01-25:
Please open PIO IDE Terminal and type this command:
```
pio device list --mdns --json-output
```

Please share an output here. Thanks!

ivankravets, 2018-01-27:
@robbor1111 @baggior

Please open PIO IDE Terminal and type this command:
```
pio device list --mdns --json-output
```

Please share an output here. Thanks!

baggior, 2018-01-28:
Exception in thread zeroconf-ServiceBrowser__services._dns-sd._udp.local.:
Traceback (most recent call last):
  File "C:\Users\Marco\.platformio\python27\Lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Users\Marco\.platformio\packages\contrib-pysite\zeroconf.py", line 1328, in run
    self.zc.send(out)
  File "C:\Users\Marco\.platformio\packages\contrib-pysite\zeroconf.py", line 2008, in send
    packet = out.packet()
  File "C:\Users\Marco\.platformio\packages\contrib-pysite\zeroconf.py", line 989, in packet
    overrun_answers += self.write_record(answer, time_)
  File "C:\Users\Marco\.platformio\packages\contrib-pysite\zeroconf.py", line 961, in write_record
    record.write(self)
  File "C:\Users\Marco\.platformio\packages\contrib-pysite\zeroconf.py", line 531, in write
    out.write_name(self.alias)
  File "C:\Users\Marco\.platformio\packages\contrib-pysite\zeroconf.py", line 914, in write_name
    count += 1
UnboundLocalError: local variable 'count' referenced before assignment

[{"ip": "192.168.1.244", "properties": {"tcp_check": "no", "ssh_upload": "no", "board": "Espressif ESP32 Dev Module", "auth_upload": "no"}, "type": "_arduino._tcp.local.", "name": "esp-async._arduino._tcp.local.", "port": 3232}]

ivankravets, 2018-01-30:
Please file this issue in https://github.com/jstasiak/python-zeroconf/issues

ivankravets, 2018-01-30:
@Fameing could you provide an output from https://github.com/platformio/platformio-core/issues/1310#issuecomment-361020798 ?

commit ba0e87b9 (2018-02-20): Fix issue with mDNS lookup service // Resolve #1310
closed 2018-02-20 by ivankravets