platformio/platformio-core #1311: Home: Libraries: Updates

sapdemon, 2018-01-25:
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 48 (char 4 - 273)

ivankravets, 2018-02-03:
Please restart VSCode. Should work now.

ivankravets, 2018-02-03:
Please open PIO IDE Terminal and type `pio home --port 888`. Run update again.

Do you see any errors?

sapdemon, 2018-02-03:
I have error "PlatformIO: IntelliSense Index failed: Error: Traceback (most recent call last):" on start PIO.

I have started "pio home --port 888" and run update.
Error is the same:
Libraries: Updates
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 48 (char 4 - 273)

sapdemon, 2018-02-03:
No, I see this error in popup window.

The result of this command (pio lib -g update -c --json-output) is empty:
[]

ivankravets, 2018-02-03:
What is your OS? Please provide a screenshot of "Home: Libraries > Installed"

sapdemon, 2018-02-03:
It is Windows 7 64x
[image]

sapdemon, 2018-02-03:
[image]

sapdemon, 2018-02-03:
[image]

ivankravets, 2018-02-05:
have you resolved this issue?

ivankravets, 2018-02-17:
@patrikom what is your OS?

patrikom, 2018-02-20:
Windows 8.1 Pro

ivankravets, 2018-02-20:
@patrikom Please open PIO IDE Terminal and type `pio home --port 8888`. Run update again.

Do you see any errors?

patrikom, 2018-02-20:
pio home --port 8888
PlatformIO Plus (https://pioplus.com) v0.14.1

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8888
|__|____|_______________________

And after that platform io home is displayed on web browser - it's OK.

> pio update
[image]

No errors.

ivankravets, 2018-02-22:
@xavcle54 what is your OS?

ivankravets, 2018-03-06:
@NRJ-Javor @pksingh2612 @ReanimationXP @kemaldede

Please open PIO IDE Terminal and type `pio home --port 8888`. Try to reproduce this issue. Do you see any errors?

kemaldede, 2018-03-07:
Sorrry i forgot to close this issue. i resolve the problem. I uninstalled the platformio and visual code studio, deleted all the files related with this programs,and i installed again. Thank you for your help

ivankravets, 2018-03-07:
Indeed, that is the issue with Fedora/CentOS distros https://bugzilla.redhat.com/show_bug.cgi?id=1147043

We have applied some fix on our side. Please `pio update` and restart IDE

ivankravets, 2018-03-12:
@aschoelzhorn What is your OS?

aschoelzhorn, 2018-03-12:
Windows 10 Pro, 64 bit
Version 1709 (build 16299.251)

ivankravets, 2018-03-12:
@aschoelzhorn that is very interesting. Could you reproduce this issue again?

ivankravets, 2018-03-16:
@cathymcintosh what is your OS?

ivankravets, 2018-06-15:
@inghardware2 could you provide an output of `python -c "import os; print os.environ"`?

Lisanderl, 2018-06-26:
{'LC_NUMERIC': 'uk_UA.UTF-8', 'MANDATORY_PATH': '/usr/share/gconf/cinnamon.mandatory.path', 'GI_TYPELIB_PATH': '/app/lib/girepository-1.0', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm-data/lisanderl', 'PLATFORMIO_CALLER': 'vscode', 'XDG_DATA_HOME': '/home/lisanderl/.var/app/com.visualstudio.code/data', 'GJS_DEBUG_OUTPUT': 'stderr', 'TERM_PROGRAM_VERSION': '1.24.1', 'XDG_CURRENT_DESKTOP': 'X-Cinnamon', 'AT_SPI_BUS_ADDRESS': 'unix:path=/run/user/1000/at-spi-bus', 'LC_PAPER': 'uk_UA.UTF-8', 'GIO_LAUNCHED_DESKTOP_FILE': '/var/lib/flatpak/exports/share/applications/com.visualstudio.code.desktop', 'LOGNAME': 'lisanderl', 'USER': 'lisanderl', 'HOME': '/home/lisanderl', 'XDG_VTNR': '7', 'PATH': '/home/lisanderl/.platformio/penv/bin:/home/lisanderl/.platformio/penv:/app/bin:/usr/bin:/home/lisanderl/.var/app/com.visualstudio.code/data/node_modules/bin', 'VSCODE_IPC_HOOK': '/run/user/1000/vscode-a651c769-1.24.1-main.sock', 'DISPLAY': ':99.0', 'TERM_PROGRAM': 'vscode', 'LANG': 'en_US.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/sh', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session0', 'QT_STYLE_OVERRIDE': 'gtk', 'LANGUAGE': 'en_US:en', 'SESSION_MANAGER': 'local/lisanderl-System-Product-Name:@/tmp/.ICE-unix/1093,unix/lisanderl-System-Product-Name:/tmp/.ICE-unix/1093', 'LC_MONETARY': 'uk_UA.UTF-8', 'QT_LINUX_ACCESSIBILITY_ALWAYS_ON': '1', 'QT_QPA_PLATFORMTHEME': 'qgnomeplatform', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'QT_ACCESSIBILITY': '1', 'NPM_CONFIG_GLOBALCONFIG': '/app/etc/npmrc', 'GTK_OVERLAY_SCROLLING': '1', 'GIO_LAUNCHED_DESKTOP_FILE_PID': '4951', 'SSH_AGENT_PID': '1148', 'VSCODE_NLS_CONFIG': '{"locale":"en-us","availableLanguages":{}}', 'XDG_CONFIG_HOME': '/home/lisanderl/.var/app/com.visualstudio.code/config', 'SHLVL': '2', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'GST_PLUGIN_SYSTEM_PATH': '/app/lib/gstreamer-1.0:/usr/lib/extensions/gstreamer-1.0:/usr/lib/gstreamer-1.0', 'XDG_RUNTIME_DIR': '/run/user/1000', 'LC_ADDRESS': 'uk_UA.UTF-8', 'XDG_CACHE_HOME': '/home/lisanderl/.var/app/com.visualstudio.code/cache', 'VSCODE_CLI': '1', 'CINNAMON_VERSION': '3.6.7', 'GDMSESSION': 'cinnamon', 'LC_NAME': 'uk_UA.UTF-8', 'PULSE_SERVER': 'unix:/run/user/1000/pulse/native', 'XDG_DATA_DIRS': '/app/share:/usr/share:/usr/share/runtime/share:/run/host/share', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'XDG_SESSION_ID': 'c1', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', '_': '/home/lisanderl/.platformio/penv/bin/python', 'DEFAULTS_PATH': '/usr/share/gconf/cinnamon.default.path', 'LC_IDENTIFICATION': 'uk_UA.UTF-8', 'DESKTOP_SESSION': 'cinnamon', 'JAVA_HOME': '/usr/lib/jvm/java-8-openjdk-amd64/bin/java', 'XDG_CONFIG_DIRS': '/app/etc/xdg:/etc/xdg', 'GTK_MODULES': 'gail:atk-bridge', 'XDG_SESSION_TYPE': 'x11', 'GDM_LANG': 'en_US', 'LC_TELEPHONE': 'uk_UA.UTF-8', 'XAUTHORITY': '/run/user/1000/Xauthority', 'LC_MEASUREMENT': 'uk_UA.UTF-8', 'PWD': '/home/lisanderl/Documents/PlatformIO/Projects/BimoSpider', 'DESKTOP_STARTUP_ID': 'cinnamon-1474-lisanderl-System-Product-Name-/usr/bin/flatpak-2_TIME3576079', 'ELECTRON_NO_ATTACH_CONSOLE': '1', 'XDG_SESSION_DESKTOP': 'cinnamon', 'PULSE_CLIENTCONFIG': '/run/user/1000/pulse/config', 'GJS_DEBUG_TOPICS': 'JS ERROR;JS LOG', 'XDG_SEAT': 'seat0'}

Lisanderl, 2018-06-26:
v3.5.3 working correct, but only by command line.

ivankravets, 2019-05-23:
Should be fixed with the latest version.

closed 2019-05-23 by ivankravets