platformio/platformio-core #1313: Projects without a specified board can't be uploaded

JackDn, 2018-01-26:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows

**PlatformIO Version** (`platformio --version`): `PlatformIO, version 3.5.2a3` (**development build**)

### Description of problem

Projects which do not specify `board` in platformio.ini can be built but not uploaded.
This was not the case in previous versions.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:avr]
platform = atmelavr
board_mcu = atmega644p
board_f_cpu = 16000000UL
upload_protocol = arduino
```

### Output

pio run -v --target upload
```
System: ATMEGA644P 16MHz
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
Modes: Finder/chain Compatibility/light
Collected 0 compatible libraries
Scanning dependencies...
No dependencies
Configuring upload protocol...
Error: BoardConfig: Board is not defined
*** [upload] Explicit exit, status 1
```

ivankravets, 2018-01-26:
Thanks a lot for the report! Please `pio upgrade --dev`

closed 2018-01-26 by ivankravets

thinkyhead, 2018-02-14:
How may we avoid this issue when using PlatformIO from Travis CI? We're getting PlatformIO via this method:
```
  - pip install -U https://github.com/platformio/platformio-core/archive/develop.zip
```

ivankravets, 2018-02-14:
@thinkyhead PIO Core 3.5.2 is planned for release next week. A temporary solution is to use line which you described above or short version:
```
- pip install platformio
- platformio upgrade --dev
```

thinkyhead, 2018-02-14:
Oddly, the line described above was the one failing. Anyway, adding a `board` fields works around the issue, so… till next week!

commit 59fb4b10 (2018-01-26): Fix issue when project without a specified board can't be uploaded // Resolve #1313