platformio/platformio-core #1318: CorePackageManager: Installing tool-scons

Commandoser, 2018-01-28:
CorePackageManager: Installing tool-scons @ ~3.20501.2
Downloading  [#-----------------------------------]    3%
Error: Please read http://bit.ly/package-manager-ioerror
Warning! Package Mirror: [Errno 0] Error
Looking for other mirror...
Error: Could not install 'tool-scons' with version requirements '~3.20501.2' for your system 'windows_x86'.
 If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

Commandoser, 2018-01-28:
Reinstalling vscode and removing ./platfomrio didn't help.

ivankravets, 2018-01-29:
Have you tried solutions from http://bit.ly/package-manager-ioerror?

Commandoser, 2018-01-29:
Thanks. First I tried to install via vscode terminal - but using standard terminal the problem was solved.

closed 2018-01-29 by Commandoser