platformio/platformio-core #1319: Feature request: Add Pycom LoPy Board

cyberman54, 2018-01-28:
- [X] Feature Request

Add [Pycom LoPy Board](https://pycom.io/product/lopy/) based on ESP32.

ivankravets, 2018-02-01:
Depends on https://github.com/platformio/platformio-core/issues/728

cyberman54, 2018-02-01:
does not necessary depend on implementation of micropython. The LoPy boards can be erased and used as native ESP32 board, for programming in C++ on platforms ESP-IDF or ardunio-espressif

d-Pixie, 2018-12-05:
Exactly, I was looking for this board here since I want to be able to use platformio to push C code to my LoPy4 😄 Python support not needed, there are already good integration with Atom/VSCode for that toolchain from the developers of the board.

ivankravets, 2019-07-23:
http://docs.platformio.org/en/latest/boards/espressif32/lopy.html

closed 2019-07-23 by ivankravets