platformio/platformio-core #1323: pio init makes QtCreator not understand Arduino C++ properly

hongquan, 2018-01-31:
What kind of issue is this?

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Ubuntu 17.10

**PlatformIO Version** (`platformio --version`): 3.5.2a5

### Description of problem

#### Steps to Reproduce

1. Have an Arduino project
2. Run `pio init --ide qtcreator --board esp07`
3. Open the .ino file in QtCreator

### Actual Results
.pro file contains `DEFINES` lines like this:

```pro
DEFINES += ARDUINO=10600
DEFINES += LWIP_OPEN_SRC
```
and QtCreator shows some errors in .ino: "expected a declaration".

### Expected Results

There should be no "expected a declaration" error.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp07]
platform = espressif8266
board = esp07
framework = arduino
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info
This is the result of attempt to fix #1299.

Before the fix (with the trailing "#" being stripped):

[image]

After the fix:

[image]

I used to proposed a fix (#1290), in which I keep the quote (`"`) and removed `#` from `DEFINES`, but unfortunately, it was rejected.

closed 2018-02-05 by ivankravets

ivankravets, 2018-02-05:
Please re-test with `pio upgrade --dev` and `pio init`

hongquan, 2018-02-06:
In 53afdc5, I see that you are doing:

```tpl
% tokens = define.split("##", 1)
DEFINES += "{{tokens[0].strip()}}"
```

but what happen if some lines use single `#`?

ivankravets, 2018-02-06:
The problem was caused by internal macros of GCC compiler. We've started to add them recently. There is no `#` in macro, see https://stackoverflow.com/questions/21891784/what-does-mean-in-c

commit 53afdc5e (2018-02-05): Fix project generator for Qt Creator IDE // Resolve #1303 Resolve #1323