platformio/platformio-core #1324: ArduCAM ESP8266 and ESP32 UNO board support

blueskyinterfaces, 2018-01-31:
X Feature Request.
I've copied this from other  #requests.

Configuration
Operating system: MacOS/VS Code

PlatformIO Version (platformio --version): 3.5.2a7

Description of problem
I would like to be able to use the Arducam ESP8266 and ESP32 Uno boards, which have settings for the Arduino IDE, but not for Platformio.

Additional info
Arducam repos: 
https://github.com/ArduCAM/ArduCAM_ESP8266_UNO
https://github.com/ArduCAM/ArduCAM_ESP32S_UNO

ivankravets, 2018-02-01:
Have you tried this library http://platformio.org/lib/show/2094/ArduCAM in pair with "NodeMCU v2" board (as a manifest)?

blueskyinterfaces, 2018-02-02:
It's not actually support for the Arducam Camera I needed. They make their own esp8266 & esp32 UNO shaped boards, and that's what I am working with. I was able to get these working with "esp12e" & "hornbilldev32" boards, respectively, though I haven't tested past a basic hello world serial print.

ivankravets, 2018-03-06:
Please file a feature request here https://github.com/platformio/platform-espressif8266/issues

closed 2018-03-06 by ivankravets