platformio/platformio-core #1325: Board Definition Request: Teensy 3.2 with SWD wires attached

atestani, 2018-02-01:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win7 and Win10

**PlatformIO Version** (`platformio --version`): version 3.5.2a7

### Description of problem
Feature request:  Teensy 3.5 and 3.6 are (presumably) able to be debugged as there is provision to add wires for SWD.  It is very easy to add these wires to Teensy 3.2.    It seems the current board definitions for 3.2 and 3.5 could be merged to add the debugging capability.

Would you please provide a Teensy 3.2 -SWD board?

ivankravets, 2018-02-03:
Moved to https://github.com/platformio/platform-teensy/issues/33

closed 2018-02-03 by ivankravets

atestani, 2018-02-04:
> Moved to platformio/platform-teensy#33

Hi Ivan

Looks like I must have posted the feature request in the wrong place,
Sorry about that.

The request is still open, right?

Thanks

Al