platformio/platformio-core #1326: Board Support Request: Z-Uno (Z-Wave Arduino prototyping board)

kreucher, 2018-02-01:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [X] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

Z-Uno is a Arduino compatible board with built in Z-Wave support, details on the board itself are here:
https://z-uno.z-wave.me/technical/

They currently support the Arduino IDE:
https://z-uno.z-wave.me/install

PJTS, 2018-07-06:
This would be a great addition, having to go back to Arduino IDE 1.6.5 isnt a great experience.

seanybaggins, 2019-03-09:
Don't know if this should be in a separate issue but debug support for the Arduino Uno would be nice as well.

samturner3, 2020-02-07:
This has been open for two years, any updates?

Mierse, 2020-03-30:
I'm also interested in support for that board. Is there any plan for?

PeterOsterdahl, 2020-09-27:
+10 for this

PoltoS, 2023-01-10:
@ivankravets Could you please guide us on how to add support of Z-Uno 2 in PlatformIO? Can we do it ourselves? Z-Uno 2 already has the support of Arduino IDE and VS Code.

PoltoS, 2024-02-24:
Of course, it will fail on ESPHome if the latter expects ESP-specific (toolchain-specific headers and libs). So you certainly need to port it to raw C or add those libs and headers as a lib in Z-Uno toolchain.

zme_make has no big secrets inside: it uses CLang to compile things and adds parsing of some custom macros. And knows how to burn the result. Your problem is not the zme_make, but a library/project that depends on toolchain-specific things.

robin-thoni, 2024-03-03:
> zme_make has no big secrets inside
Not sure about that. It seems to magically, conditionally add some (LLCore, etc) libraires at build time. Also, it seems to fail to build when using something as simple as `std::vector` :/

PoltoS, 2024-03-03:
Please provide the code you have problems with and our developers will look at it

robin-thoni, 2024-03-03:
I didn't realize you were part of Z-Wave.me. Is there any chance you, as Z-Wave.me as a whole, to work on pio support? Especially since Home Assistant has now joined the Z-Wave Alliance, you would get *WAY* more visibility (and sells) if ESPHome would support it (https://github.com/esphome/feature-requests/issues/1839).
Making a Z-Wave custom device in seconds, like w/ ESP32 is a killer feature. Now Zigbee is built in new ESP32 chips, that's what's gonna happen for Zigbee. But IMHO, Z-Wave > Zigbee :)

For the vector issue, just `#include <vector>` makes it crash:

```
#include <vector>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

```
/home/robin/tools/arduino-1.8.15/arduino-builder -dump-prefs -logger=machine -hardware /home/robin/tools/arduino-1.8.15/hardware -hardware /home/robin/.arduino15/packages -tools /home/robin/tools/arduino-1.8.15/tools-builder -tools /home/robin/tools/arduino-1.8.15/hardware/tools/avr -tools /home/robin/.arduino15/packages -built-in-libraries /home/robin/tools/arduino-1.8.15/libraries -libraries /home/robin/Arduino/libraries -fqbn=Z-Uno2:zw_cm4f:zuno2:Frequency=Eu,TXPower=Normal,RFLogging=Off,CleanNVM=Off,UpdateLicense=On -ide-version=10815 -build-path /tmp/arduino_build_52682 -warnings=none -build-cache /tmp/arduino_cache_948672 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.zme_make.path=/home/robin/.arduino15/packages/Z-Uno2/tools/zme_make/0.4.1 -prefs=runtime.tools.zme_make-0.4.1.path=/home/robin/.arduino15/packages/Z-Uno2/tools/zme_make/0.4.1 -prefs=runtime.tools.libClang.path=/home/robin/.arduino15/packages/Z-Uno2/tools/libClang/11.0.1 -prefs=runtime.tools.libClang-11.0.1.path=/home/robin/.arduino15/packages/Z-Uno2/tools/libClang/11.0.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/robin/.arduino15/packages/Z-Uno2/tools/arm-none-eabi-gcc/7.2.4 -prefs=runtime.tools.arm-none-eabi-gcc-7.2.4.path=/home/robin/.arduino15/packages/Z-Uno2/tools/arm-none-eabi-gcc/7.2.4 -verbose /tmp/arduino_modified_sketch_540012/sketch_mar03a.ino
/home/robin/tools/arduino-1.8.15/arduino-builder -compile -logger=machine -hardware /home/robin/tools/arduino-1.8.15/hardware -hardware /home/robin/.arduino15/packages -tools /home/robin/tools/arduino-1.8.15/tools-builder -tools /home/robin/tools/arduino-1.8.15/hardware/tools/avr -tools /home/robin/.arduino15/packages -built-in-libraries /home/robin/tools/arduino-1.8.15/libraries -libraries /home/robin/Arduino/libraries -fqbn=Z-Uno2:zw_cm4f:zuno2:Frequency=Eu,TXPower=Normal,RFLogging=Off,CleanNVM=Off,UpdateLicense=On -ide-version=10815 -build-path /tmp/arduino_build_52682 -warnings=none -build-cache /tmp/arduino_cache_948672 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.zme_make.path=/home/robin/.arduino15/packages/Z-Uno2/tools/zme_make/0.4.1 -prefs=runtime.tools.zme_make-0.4.1.path=/home/robin/.arduino15/packages/Z-Uno2/tools/zme_make/0.4.1 -prefs=runtime.tools.libClang.path=/home/robin/.arduino15/packages/Z-Uno2/tools/libClang/11.0.1 -prefs=runtime.tools.libClang-11.0.1.path=/home/robin/.arduino15/packages/Z-Uno2/tools/libClang/11.0.1 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/robin/.arduino15/packages/Z-Uno2/tools/arm-none-eabi-gcc/7.2.4 -prefs=runtime.tools.arm-none-eabi-gcc-7.2.4.path=/home/robin/.arduino15/packages/Z-Uno2/tools/arm-none-eabi-gcc/7.2.4 -verbose /tmp/arduino_modified_sketch_540012/sketch_mar03a.ino
Using board 'zuno2' from platform in folder: /home/robin/.arduino15/packages/Z-Uno2/hardware/zw_cm4f/3.0.10
Using core '/' from platform in folder: /home/robin/.arduino15/packages/Z-Uno2/hardware/zw_cm4f/3.0.10
Detecting libraries used...
/home/robin/.arduino15/packages/Z-Uno2/tools/zme_make/0.4.1/zme_make arduino_preproc /tmp/arduino_build_52682/sketch/sketch_mar03a.ino.cpp /home/robin/.arduino15/packages/Z-Uno2/hardware/zw_cm4f/3.0.10/cores
Found 1.8.x project structure (File:sketch_mar03a.ino.cpp). Converting it...
Generating function prototypes...
/home/robin/.arduino15/packages/Z-Uno2/tools/zme_make/0.4.1/zme_make arduino_preproc /tmp/arduino_build_52682/sketch/sketch_mar03a.ino.cpp /home/robin/.arduino15/packages/Z-Uno2/hardware/zw_cm4f/3.0.10/cores
Found 1.8.x project structure (File:sketch_mar03a.ino.cpp). Converting it...
          INCREMENTAL PREPROC. The sketch is the same.
/home/robin/tools/arduino-1.8.15/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_52682/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/robin/.arduino15/packages/Z-Uno2/tools/zme_make/0.4.1/zme_make build /tmp/arduino_build_52682/sketch_mar03a.ino -S /home/robin/.arduino15/packages/Z-Uno2/hardware/zw_cm4f/3.0.10/cores -S /home/robin/.arduino15/packages/Z-Uno2/hardware/zw_cm4f/3.0.10/libraries -S /home/robin/.arduino15/packages/Z-Uno2/tools/arm-none-eabi-gcc/7.2.4/lib/gcc/arm-none-eabi/7.2.1/include -B /tmp/arduino_build_52682 -T /home/robin/.arduino15/packages/Z-Uno2/tools/arm-none-eabi-gcc/7.2.4/bin -O arduino_ide -lcl /home/robin/.arduino15/packages/Z-Uno2/tools/libClang/11.0.1 -O BO:-DARDUINO=152 -O BO:-DARDUINO_ARCH_ZUNOG2 -O BO:-DZME_ZUNOG2 -O LO:-lm
          ***** Building sketch:/tmp/arduino_build_52682/sketch_mar03a.ino *****
          -----------------------------------------------------------------------------------------------
[... cut ...]
stl_algobase.h:270:7: error: expected unqualified-id before 'return'
       return __a;
       ^~~~~~
stl_algobase.h:271:5: error: expected declaration before '}' token
     }
     ^

exit status 10
/home/robin/tools/arduino-1.8.15/arduino-builder returned 10
Error compiling for board Z-Wave>ME Z-Uno2.

```

Board is `Z-Uno 2`, programmer is `Z-Uno-2G Programmer`, using Arduiono IDE. I can repro the same with my custom pio toolchain.

I also tried to run it w/ `arm-none-eabi-gcc` `7.2.4` coming from pio (in case you would have tweaked it), same result

PoltoS, 2024-03-03:
Okay, I see the issue. For whatever reason vector relies on min and max with three arguments, which is insane. We will check it and fix. And __comp is missing or wrong.

@p0lyg0n1

Frankly speaking, it is hard for us to do pio support as we never used it.  But we can actively assist if you could guide us.

On the other side, we have our new Z-Uno configurator just released. Currently it is for the Z-Uno Shield only, but we hope to add the simple Z-Uno too.

It is only clicks to get your sketch generated and even uploaded right from the browser - no programming, no Arduino IDE, no soldering. Easy as 1-2-3.

Check it at https://z-uno.z-wave.me/configurator/#/shield

robin-thoni, 2024-04-10:
> We will check it and fix. And __comp is missing or wrong.

@PoltoS Is there any news in this? Sounds like a big blocker for PIO adoption

PoltoS, 2024-04-10:
Hello! Sorry for the delay. We checked it and your compilation line looks weired.

1. Please try to use the latest beta package from our site.
2. Take the GCC provided in the package.
3. Your line contains /home/robin/.arduino15/packages/Z-Uno2/tools/arm-none-eabi-gcc/**7.2.4**/lib/gcc/arm-none-eabi/**7.2.1**/include - there should be no 7.2.1 in 7.2.4. Looks you messed it up.
4. Take the compilation line from platform.txt: `recipe.hooks.sketch.prebuild.1.pattern = "{tools.zprog.cmd.path}" build "{build.path}/{build.project_name}" -S "{tools.zprog.cmd.cores}" -S "{tools.zprog.cmd.libraries}"  -S "{tools.zprog.cmd.include}" -B  "{build.path}" -T "{tools.zprog.cmd.bin}" -O "arduino_ide" -lcl "{tools.zprog.cmd.clang_lib}" -O BO:-DARDUINO=152 -O BO:-DARDUINO_ARCH_ZUNOG2 -O BO:-DZME_ZUNOG2 -O LO:-lm -C "{build.chiptype}" `
(in your case the problem is in `{tools.zprog.cmd.include}`)
5. In the latest beta we use gcc 10.3.1: `tools.zprog.cmd.include={runtime.tools.arm-none-eabi-gcc.path}/lib/gcc/arm-none-eabi/10.3.1/include` - please try it.

Hope this helps!

PoltoS, 2024-10-31:
FYI, we now have an easy-to-use online compilation tool, so you don't even need to install anything, just take any sketch on our https://z-uno.z-wave.me site and press the yellow **Upload to Z-Uno** button. You can also use an online editor https://z-uno.z-wave.me/editor/ or an online configurator (for Z-Uno 2 Shield): https://z-uno.z-wave.me/configurator/#/shield

robin-thoni, 2024-10-31:
@nomailme I completely missed @PoltoS reply, and didn't give it another try yet

@PoltoS

> Please try to use the latest beta package from our site.

I'll try to use the current latest/beta and see what I can do w/ it.

> there should be no 7.2.1 in 7.2.4. Looks you messed it up.

I don't think I messed w/ that config, but who knows... I'll try again.

> FYI, we now have an easy-to-use online compilation tool

Sure, but that only handles extremely basic use-cases, with little to no support for external hardware.

On the other side, ESPHome (through PIO) has a quite good support for a lot of peripherals, and allows to create automations on the device itself w/ only a few lines of YAML.

Having an official support for ESPHome would be beneficial to both end-users, and Z-Wave.me, since you'd benefit from ESPHome community and the free advertisement of their [supported hardware page](https://esphome.io/#supported-microcontrollers).

PoltoS, 2024-10-31:
We are open to cooperate and help. Please let us know if you need any assistance, I'll try to push it internally.

robin-thoni, 2024-10-31:
Understanding how `zme_make` works internally would help a lot. I'm currently getting an infinite recursion when adding the `esphome` folder in the source list:
```
INFO ESPHome 2024.11.0-dev
INFO Reading configuration config/dev-zuno01.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing dev-zuno01 (board: zuno2; framework: arduino; platform: zuno)
--------------------------------------------------------------------------------
/esphome/.temp/platformio/packages/toolchain-zme_make/zme_make build src/main.cpp -S /esphome/.temp/platformio/packages/framework-zuno2/cores -S /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome -B /workspaces/esphome/config/.esphome/build/dev-zuno01/.pioenvs/dev-zuno01 -T /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/bin -lcl /esphome/.temp/platformio/packages/toolchain-libclang
          ***** Building sketch:/workspaces/esphome/config/.esphome/build/dev-zuno01/src/main.cpp *****
          -----------------------------------------------------------------------------------------------
SOURCE LIST:['/workspaces/esphome/config/.esphome/build/dev-zuno01/src/', '/workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/', '/workspaces/esphome/config/.esphome/build/dev-zuno01/.pioenvs/dev-zuno01/main/', '/workspaces/esphome/config/.esphome/build/dev-zuno01/.pioenvs/dev-zuno01/main/zuno_preproc/', '/esphome/.temp/platformio/packages/framework-zuno2/cores/', '/esphome/.temp/platformio/packages/framework-zuno2/cores/includes/', '/esphome/.temp/platformio/packages/framework-zuno2/cores/ZWSupport/', '/esphome/.temp/platformio/packages/framework-zuno2/cores/emlib/', '/esphome/.temp/platformio/packages/framework-zuno2/cores/libft/', '/esphome/.temp/platformio/packages/framework-zuno2/cores/LLCore/', '/workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/', '/workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/core/', '/workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/components/']

preprocessing "main.cpp"                 ..............................
preprocessing "main.cpp"                 ..............................                            OK
          Core version: 03.10 Channels:0

Gathering project files                  ..............................
Gathering project files                  ..............................                            OK

compiling "zwaveme_mktime.c"             ..............................
compiling "zwaveme_mktime.c"             ..............................                        FAILED
          In file included from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/core/time.h:5:0,
                           from /esphome/.temp/platformio/packages/framework-zuno2/cores/libft/zwaveme_mktime.c:1:
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:60:11: error: '::clock_t' has not been declared
             using ::clock_t;
                     ^~~~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:61:11: error: '::time_t' has not been declared
             using ::time_t;
                     ^~~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:62:11: error: '::tm' has not been declared
             using ::tm;
                     ^~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:64:11: error: '::clock' has not been declared
             using ::clock;
                     ^~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:65:11: error: '::difftime' has not been declared
             using ::difftime;
                     ^~~~~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:66:11: error: '::mktime' has not been declared
             using ::mktime;
                     ^~~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:67:11: error: '::time' has not been declared
             using ::time;
                     ^~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:68:11: error: '::asctime' has not been declared
             using ::asctime;
                     ^~~~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:69:11: error: '::ctime' has not been declared
             using ::ctime;
                     ^~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:70:11: error: '::gmtime' has not been declared
             using ::gmtime;
                     ^~~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:71:11: error: '::localtime' has not been declared
             using ::localtime;
                     ^~~~~~~~~
          /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/ctime:72:11: error: '::strftime' has not been declared
             using ::strftime;
                     ^~~~~~~~
          In file included from /esphome/.temp/platformio/packages/framework-zuno2/cores/libft/zwaveme_mktime.c:1:0:
[... cut ...]
Gathering project files                  ..............................                            OK

Error  0	 Can't find ld-file for project!
*** [.pioenvs/dev-zuno01/main/main_ino_signed.bin] Error 235
```

Here's my current attempt at creating a PIO platform: https://github.com/robin-thoni/pio-platform-zuno

Also, I couldn't find the 10.x gcc toolchain you mentioned in the [json file](https://z-uno.z-wave.me/files/z-uno2/package_z-wave2.me_index.json). I can only see `arm-none-eabi-gcc-7_2_4-linux64`.

Can you help on this? THanks!

PoltoS, 2024-10-31:
Please use the beta 3.13 (https://z-uno.z-wave.me/files/z-uno2/package_z-wave2.me_beta_index.json).

amatilda, 2024-10-31:
@robin-thoni 
Hi. What are you trying to build with z-uno?

nomailme, 2024-11-01:
> FYI, we now have an easy-to-use online compilation tool, so you don't even need to install anything, just take any sketch on our https://z-uno.z-wave.me site and press the yellow **Upload to Z-Uno** [...]

I second @robin-thoni thoughts on having a code as a project. I guess I'm a bit old-fashioned but I prefer having my projects defined as code, split in classes, stored in a git, with unit tests that are triggered by CI.

robin-thoni, 2024-11-04:
Sounds like I'm getting similar results w/ the latest beta:
```
INFO ESPHome 2024.11.0-dev
INFO Reading configuration config/dev-zuno01.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing dev-zuno01 (board: zuno2; framework: arduino; platform: zuno)
--------------------------------------------------------------------------------
/esphome/.temp/platformio/packages/toolchain-zme_make@1.501.0/zme_make build src/main.cpp -S /esphome/.temp/platformio/packages/framework-zuno2@1.30013.0/cores -S /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome -B /workspaces/esphome/config/.esphome/build/dev-zuno01/.pioenvs/dev-zuno01 -T /esphome/.temp/platformio/packages/toolchain-gccarmnoneeabi@1.100301.0/bin -lcl /esphome/.temp/platformio/packages/toolchain-libclang@1.110001.0
          ***** Building sketch:/workspaces/esphome/config/.esphome/build/dev-zuno01/src/main.cpp *****
          -----------------------------------------------------------------------------------------------

preprocessing "main.cpp"                 ..............................
preprocessing "main.cpp"                 ..............................                            OK
          Core version: 03.13 Channels:0
          Chip depending configuration. Chip:ZGM130S037HGN1

Gathering project files                  ..............................
Gathering project files                  ..............................                            OK

compiling "main_ino.cpp"                 ..............................
compiling "main_ino.cpp"                 ..............................                        FAILED
          In file included from /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/bits/hashtable.h:35,
                           from /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/unordered_map:46,
                           from /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/functional:61,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/core/component.h:5,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/core/application.h:5,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome.h:3,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/main.cpp:3:
          /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/bits/hashtable_policy.h:560:47: error: macro "max" requires 2 arguments, but only 1 given
            560 |  _M_next_resize = numeric_limits<size_t>::max();
                |                                               ^
          In file included from /workspaces/esphome/config/.esphome/build/dev-zuno01/.pioenvs/dev-zuno01/main//main_ino.cpp:10:
          /esphome/.temp/platformio/packages/framework-zuno2@1.30013.0/cores/includes/Arduino.h:67: note: macro "max" defined here
             67 | #define max(a,b)     ((a)>(b)?(a):(b))
                | 
          In file included from /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/bits/stl_algo.h:60,
                           from /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/functional:65,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/core/component.h:5,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/core/application.h:5,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome.h:3,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/main.cpp:3:
          /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/bits/algorithmfwd.h:412:41: error: macro "max" passed 3 arguments, but takes just 2
            412 |     max(const _Tp&, const _Tp&, _Compare);
                |                                         ^
          In file included from /workspaces/esphome/config/.esphome/build/dev-zuno01/.pioenvs/dev-zuno01/main//main_ino.cpp:10:
          /esphome/.temp/platformio/packages/framework-zuno2@1.30013.0/cores/includes/Arduino.h:67: note: macro "max" defined here
             67 | #define max(a,b)     ((a)>(b)?(a):(b))
                | 
          In file included from /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/bits/stl_algo.h:60,
                           from /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/functional:65,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/core/component.h:5,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome/core/application.h:5,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/esphome.h:3,
                           from /workspaces/esphome/config/.esphome/build/dev-zuno01/src/main.cpp:3:
          /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/bits/algorithmfwd.h:425:41: error: macro "min" passed 3 arguments, but takes just 2
[... cut ...]
          /workspaces/esphome/config/toolchain-gccarmnoneeabi@1.100301.0/arm-none-eabi/include/c++/10.3.1/bits/uniform_int_dist.h:129:56: error: macro "max" requires 2 arguments, but only 1 given
            129 |           _IntType __b = numeric_limits<_IntType>::max())
                |                                                        ^
          In file included from /workspaces/esphome/config/.esphome/build/dev-zuno01/.pioenvs/dev-zuno01/main//main_ino.cpp:10:
          /esphome/.temp/platformio/packages/framework-zuno2@1.30013.0/cores/includes/Arduino.h:67: note: macro "max" defined here
             67 | #define max(a,b)     ((a)>(b)?(a):(b))
                | 
```

PoltoS, 2024-11-05:
`max` macro is conflicting. You can do `#undefine max` just before `#include "esphome.h"`