platformio/platformio-core #1335: Home: Could not load installed platforms

LazosNalp, 2018-02-05:
PIO Core Call Error: Upon execvpe /home/lazos/.platformio/penv/bin/platformio ['/home/lazos/.platformio/penv/bin/platformio', 'platform', 'list', '--json-output'] in environment id 140079493572880
:Traceback (most recent call last):
  File "/home/lazos/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 445, in _fork
    environment)
  File "/home/lazos/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 523, in _execChild
    os.execvpe(executable, args, environment)
  File "/usr/lib/python2.7/os.py", line 355, in execvpe
    _execvpe(file, args, env)
  File "/usr/lib/python2.7/os.py", line 370, in _execvpe
    func(file, *argrest)
OSError: [Errno 2] No such file or directory

ivankravets, 2018-02-06:
Have you modified Python installation? try to remove `/home/lazos/.platformio` folder and repeat.

LazosNalp, 2018-02-06:
Thank you for reply. I've already fix it removing  /home/lazos/.platformio.

closed 2018-02-06 by LazosNalp